carla-simulator/carla #4878: Build system: make launch errors in completely untouched classes

roque2205, 2021-11-12:
Hi guys,

I have been through quite a hustle trying to implement an own radar sensor and I wanted to share my story with you.

1. I looked at the sensor tutorial and did the same steps but created my own class "AGroundTruthRadar". Outcome: PythonAPI compiles without errors, launch doesn't throw errors in my files but in carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp.
2. I took the sensor tutorial and only changed the files' contents to my needs. My sensor remains named "ASafeDistanceSensor" for pure belief in magic. Outcome: PythonAPI compiles without errors, launch compiles without errors.
3. I change the class name to "AGroundTruthRadar". Outcome: PythonAPI compiles without errors, launch doesn't throw errors in my files but in carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp.
4. I freak out and change the name back to "ASafeDistanceSensor". Outcome: PythonAPI compiles without errors, launch doesn't throw errors in my files but in carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp.
5. I try compiling with the same source code on another machine. Outcome: PythonAPI compiles without errors, launch doesn't throw errors in my files but in "carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Util\BoundingBoxCalculator.cpp".

Please tell me there is a solution for this...

Best regards
Robert

benraziel, 2021-11-13:
I've encountered a similar issue, I believe the root cause is Carla's use of unity builds. see https://github.com/carla-simulator/carla/issues/4225 for more details.
unity builds may hide missing #include declarations, since multiple .cpp files are concatenated into a single .cpp file by the build system. Adding new files changes the grouping of .cpp files in the build, and thus can "uncover" those missing #include declarations.
The current workaround we've found is to add the missing #include's manually, whenever we encounter compilation errors.

uservinu, 2022-01-12:
I am facing the same issue after creating a new c++ actor
`E:/develop/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp(28): error C2027: use of undefined type 'carla::road::element::RoadInfoSignal'
  E:\develop\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Traffic\SignComponent.h(23): note: see declaration of 'carla::road::element::RoadInfoSignal'
  E:/develop/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp(28): error C2530: 'validity': references must be initialized
  E:/develop/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp(28): error C3531: 'validity': a symbol whose type contains 'auto' must have an initializer
  E:/develop/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp(28): error C2143: syntax error: missing ';' before ':'
  E:/develop/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp(28): error C2143: syntax error: missing ';' before ')'
  E:/develop/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp(30): error C2660: 'carla::geom::Math::GenerateRange': function does not take 1 arguments
  E:\develop\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\include\carla/geom/Math.h(124): note: see declaration of 'carla::geom::Math::GenerateRange'
  E:/develop/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp(30): error C3531: 'lane': a symbol whose type contains 'auto' must have an initializer
  E:/develop/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp(30): error C2143: syntax error: missing ';' before ':'
  E:/develop/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp(30): error C2660: 'carla::geom::Math::GenerateRange': function does not take 1 arguments
  E:\develop\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\include\carla/geom/Math.h(124): note: see declaration of 'carla::geom::Math::GenerateRange'
  E:/develop/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp(30): error C2143: syntax error: missing ';' before ')'
  E:/develop/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp(36): error C2027: use of undefined type 'carla::road::element::RoadInfoSignal'
  E:\develop\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Traffic\SignComponent.h(23): note: see declaration of 'carla::road::element::RoadInfoSignal'`

roque2205, 2022-01-12:
Hi, I can confirm that currently the most promising approach is adding all the missing includes. It took me some time, but in the end it worked.

uservinu, 2022-01-12:
Fixed by adding missing includes in carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp

```
#include "YieldSignComponent.h"
#include "TrafficLightState.h"
#include "Carla/Vehicle/CarlaWheeledVehicle.h"
#include <queue>

#include <compiler/disable-ue4-macros.h>
#include <carla/road/element/RoadInfoSpeed.h>
#include <carla/road/element/RoadInfoSignal.h>
#include <compiler/enable-ue4-macros.h>

```

pritesh604, 2022-07-01:
```
/home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/carla/rpc/ActorAttributeType.h:11:17: error: namespaces can only be defined in global or namespace scope
namespace carla {
                ^
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.3_of_3.cpp:4:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Trigger/FrictionTrigger.cpp:6:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Trigger/FrictionTrigger.h:9:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEpisode.h:9:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorDispatcher.h:9:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorDefinition.h:9:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h:13:
/home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Carla/ActorAttribute.generated.h:21:1: error: expected expression
template<> CARLA_API UScriptStruct* StaticStruct<struct FActorAttribute>();
^
/home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Carla/ActorAttribute.generated.h:28:1: error: expected expression
template<> CARLA_API UScriptStruct* StaticStruct<struct FActorVariation>();
^
/home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Carla/ActorAttribute.generated.h:45:1: error: expected expression
template<> CARLA_API UEnum* StaticEnum<EActorAttributeType>();
^
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.3_of_3.cpp:4:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Trigger/FrictionTrigger.cpp:6:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Trigger/FrictionTrigger.h:9:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEpisode.h:9:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorDispatcher.h:9:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorDefinition.h:9:
/home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h:22:15: error: no member named 'ActorAttributeType' in namespace 'carla::rpc'; did you mean 'EActorAttributeType'?
  Bool      = CARLA_ENUM_FROM_RPC(Bool)      UMETA(DisplayName = "Bool"),
              ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h:15:51: note: expanded from macro 'CARLA_ENUM_FROM_RPC'
#define CARLA_ENUM_FROM_RPC(e) static_cast<uint8>(carla::rpc::ActorAttributeType:: e)
                                                  ^~~~~~~~~~~~
/home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h:19:12: note: 'EActorAttributeType' declared here
enum class EActorAttributeType : uint8
           ^
/home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h:22:35: error: no member named 'Bool' in 'EActorAttributeType'
  Bool      = CARLA_ENUM_FROM_RPC(Bool)      UMETA(DisplayName = "Bool"),
              ~~~~~~~~~~~~~~~~~~~~^~~~~
/home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h:15:84: note: expanded from macro 'CARLA_ENUM_FROM_RPC'
#define CARLA_ENUM_FROM_RPC(e) static_cast<uint8>(carla::rpc::ActorAttributeType:: e)
                                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
/home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h:23:15: error: no member named 'ActorAttributeType' in namespace 'carla::rpc'; did you mean 'EActorAttributeType'?
  Int       = CARLA_ENUM_FROM_RPC(Int)       UMETA(DisplayName = "Integer"),
              ^~~~~~~~~~~~~~~~~~~~~~~~
/home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h:15:51: note: expanded from macro 'CARLA_ENUM_FROM_RPC'
#define CARLA_ENUM_FROM_RPC(e) static_cast<uint8>(carla::rpc::ActorAttributeType:: e)
                                                  ^~~~~~~~~~~~
/home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h:19:12: note: 'EActorAttributeType' declared here
enum class EActorAttributeType : uint8
           ^
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.2_of_3.cpp:21:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaReplayerHelper.cpp:20:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/TrafficLightBase.h:12:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/TrafficLightComponent.h:10:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/SignComponent.h:12:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDrive/OpenDrive.h:9:
In file included from /home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDrive/OpenDriveMap.h:10:
[... cut ...]
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h:15:51: note: expanded from macro 'CARLA_ENUM_FROM_RPC'
#define CARLA_ENUM_FROM_RPC(e) static_cast<uint8>(carla::rpc::ActorAttributeType:: e)
                                                  ^~~~~~~~~~~~
```

I am also facing this issue. I have created my own sensor model in carla and  didn't touch any other files, still I am getting this error. Did anyone faced this issue?

pritesh604, 2022-07-01:
@benraziel can you please help me with this problem?