carla-simulator/carla #4886: Problem with create pedestrian jaywalking in version 0.9.12

wjx-3510886, 2021-11-15:
Hi,
I'm using carla 0.9.12 under windows 10.
Problem you have experienced:
Today I create an invisible crosswalk in the street on the original Town03 map. According to the document **https://carla.readthedocs.io/en/0.9.12/tuto_M_generate_pedestrian_navigation/** , I  made sure this invisible crosswalk mesh had the same details as other Town03 crosswalks,  I extended it to both ends of the road . And name it Road_Crosswalk_Town03_XX , then I selected all with CTRL +A and clicked Carla Exporter. Then I put both the generated obj file and town03.xodr from Carla\Maps\OpenDrive into DockerUtils\dist and run the command to place the generated bin file in the specified location.  In my tests with Generate_traffic.py, I found that pedestrians would not cross my added crosswalk. According to @bernatx  suggestting in  **https://github.com/carla-simulator/carla/issues/3086**  I Select only the tags in NAVI_pedestrian, set part of the tags to NoExport, and then try again, the result also fails. After that,I used Recast&Detour,suggest in **https://github.com/carla-simulator/carla/issues/1899**,I do the above operation, the result is also failed.I don't know how to finish jaywalking on the road. 
Glad about any help!
[image]
[image]

bernatx, 2021-11-15:
Hi,
I suppose you replace the original .bin in the Nav folder, with the name Town03.bin. 
So, what I think is happening, is that CARLA uses a cache in client for those kind of files (that are transferred from server to client when needed). What happens is that CARLA is still using the version he has in cache, and you are replacing the one in the server but not the one the client has in cache.
When you replace any file like that (pedestrians, traffic manager or openDrive) you should remove the cache folder. Just replace the **carlacache** in the user folder.
In the future we need to hash to check if the file from server is different that the cached one. Sorry for the confusion.
Regards

jackbart94, 2021-11-17:
Hey @wjx-3510886! Traffic Manager does slow down vehicles at intersections, that is true. It's because vehicles need to be able to turn correctly at intersections. Regarding accelerations, they are as realistic as possible, we just updated all vehicle's physics to match the real ones in 0.9.13. If you can, change your CARLA version to 0.9.13 and give it a try. Otherwise, you can try to work around this by using enable_constant_velocity(self, velocity)

wjx-3510886, 2021-11-19:
Thanks very much @jackbart94 . Besides that,  is there a way to create a vehicle that is not on the road (i.e. in a yard) and then drive it on the road (like a car popping out of an alley or yard)? Thank you for the work.  It's fantastic.
[image]

closed 2022-05-02 by stale[bot]