carla-simulator/carla #4890: libstdc++ version must be at least 4.8

duongtran11, 2021-11-16:
CARLA version: latest
Platform/OS: Ubuntu 20.04
Problem you have experienced: libstdc++ version must be at least 4.8
What you expected to happen: make PythonAPI
Other information (documentation you consulted, workarounds you tried): I'm building Carla Simulator from source on Ubuntu 20.04 mchine following the instruction in the docs, when I tried make PythonAPI I got this error
[image]

ls19920505, 2021-11-18:
I also encountered the same problem, did you solve it?

duongtran11, 2021-11-18:
@ls19920505 no I didn't 
I checked the file cmake/modules/CheckCompilerVersion.cmake and found out that the reason for this `libstdc++ version must be at least 4.8` error maybe cause by GLIBCXX release date is sooner than 22.03.2013. But then I checked my gcc version it's 9.3 and release date is some where in 2018
So it doesn't make any sense
Please tell me if you find any solution

MrLerry, 2022-03-07:
Any update on this?

ThankPan, 2023-02-08:
I encountered this same problem after the newly updated document about using clang-10 in Ubuntu 20.04. The error message in the CMakeError.log is as follows:
`Performing C++ SOURCE FILE Test LLVM_LIBSTDCXX_MIN failed with the following output:
Change Dir: /home/reinht/Desktop/carla/carla/Build/llvm-8.0-build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/ninja cmTC_4c96e && [1/2] Building CXX object CMakeFiles/cmTC_4c96e.dir/src.cxx.o
FAILED: CMakeFiles/cmTC_4c96e.dir/src.cxx.o 
/usr/bin/clang++-10    -DLLVM_LIBSTDCXX_MIN  -std=c++0x -o CMakeFiles/cmTC_4c96e.dir/src.cxx.o -c src.cxx
src.cxx:2:10: fatal error: 'iosfwd' file not found
#include <iosfwd>
         ^~~~~~~~
1 error generated.
ninja: build stopped: subcommand failed.

Source file was:

#include <iosfwd>
#if defined(__GLIBCXX__)
#if __GLIBCXX__ < 20130322
#error Unsupported libstdc++ version
#endif
#endif
int main() { return 0; }`

It seems that the outputted "libstdc++ version must be at least 4.8" is caused by failing the test since 'iosfwd' not found.
Is there any possible solution to solve the root problem?

ThankPan, 2023-02-09:
> I encountered this same problem after the newly updated document about using clang-10 in Ubuntu 20.04. The error message in the CMakeError.log is as follows:
>
> ```
> Performing C++ SOURCE FILE Test [...]

Simply solved it by `sudo apt install g++-n`. n is the version of GCC clang is using.

yuanhaorannnnnn, 2023-05-11:
> @ls19920505 no I didn't I checked the file cmake/modules/CheckCompilerVersion.cmake and found out that the reason for this `libstdc++ version must be at least 4.8` error maybe cause by GLIBCXX [...]

#if __GLIBCXX__ < ${GCC_MIN_DATE}
this line of code makes no sense...compare version with date?

yuanhaorannnnnn, 2023-05-11:
> GCC

g++ and gcc same version, not for clang

SoonMoom, 2024-05-15:
redownload usr/lib/x86_64 ... libstdc++.so.*** then reinstall or cp from other computer , if you remove libstdc++.so.*** system apt tec. will not work!!!

Tabula-Rasa-Oriental, 2025-01-19:
How to solve it? SOS