carla-simulator/carla #4891: scenrio_runner crashes CARLA Server while loading maps

MaddysonBlaze, 2021-11-16:
I have been trying to run a simulation on using OpenScenario files and scenario runner. Please find my local setup details below.

CARLA version: **0.9.12** precompiled version and source build version

Platform/OS: **Ubuntu 18.04 LTS**

GPU : **NVIDEA GeFORCE RTX 3070**

Graphics driver : **Nvidia driver version : 495.44**

Problem you have experienced:

I get the below console output when i run scenario_runner.py script

    scenario_runner main function
    API Version: 0.9.12
    Segmentation fault (core dumped)

And the carla server crashes with the console output as shown below

    LowLevelFatalError [File:Unknown] [Line: 803]
    VulkanRHI::vkGetPhysicalDeviceSurfaceFormatsKHR(Device.GetPhysicalHandle(), Surface, &NumFormats, nullptr) failed, VkResult=-3
    at /home/jenkins/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanSwapChain.cpp:179
    with error VK_ERROR_INITIALIZATION_FAILED
    Signal 11 caught.
    Malloc Size=65538 LargeMemoryPoolOffset=65554
    CommonUnixCrashHandler: Signal=11
    Malloc Size=131160 LargeMemoryPoolOffset=196744
    Malloc Size=131160 LargeMemoryPoolOffset=327928
    Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
    Segmentation fault (core dumped)

Steps to reproduce:
1. run carla server using the command `./CarlaUE4.sh`
2. run scenario_runner.py using the command `python scenario_runner.py --openscenario /path/to/xosc/file`

I tried launching Carla server with vulkan attribute as well : `./CarlaUE4.sh -vulkan`
But i get the same error with this as well.

I tried to follow the steps from the thread [4582](https://github.com/carla-simulator/carla/issues/4582) and [4711](https://github.com/carla-simulator/carla/issues/4711)

But there it worked on the docker version of Carla and the problem on the pre-compiled version and source build version still exists.

Please let me know if any further details are needed.

bhuvanofc, 2021-11-16:
I have the same issue while running a source build version on my Linux system.

MaddysonBlaze, 2021-11-16:
@bhuvanofc, were you able to solve the problem ?

bhuvanofc, 2021-11-17:
@MaddysonBlaze No unfortunately not. Hope to get some solution for it here

varunjammula, 2022-01-31:
Same issue when I run multiple scenarios on CARLA 0.9.13 built from source on Ubuntu 20

abdooo786, 2022-04-22:
i don't know why , but it worked with me using "./CarlaUE4.sh -opengl13" 
ubuntu 18.04
0.9.11
ryzen 5 3400G

ChunTing-Wu, 2022-07-26:
Hi gus,

I have the same issue.

ubuntu 18.04
Carla Version 0.9.13

Nvidia driver 470
RTX 3070

ahmedhatem3h, 2022-08-17:
Hello guys,

I have the same problem. First scenario runs okay, when it is finished, Carla server crashes when loading the second scenario and gives the following error:

4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

Did anyone know how to fix this?

Ubuntu 20
Carla 0.9.13
NVIDIA driver 515
RTX 2060

ghazalmatt3r, 2022-09-09:
I have the same problem...
I tried running a scenario with OpenDrive map and next time I tried running ./CarlaUE4.sh I got this error. I tried re-installing carla but it didn't help...

omitheaussie, 2023-05-12:
Has anyone found solution to this?

xiboli, 2024-02-01:
I tried to load my own map, which is built by road runner and it also gives me the same error like Malloc Size=131160 LargeMemoryPoolOffset=327928. And I find the reason. Because one junction in my map did not have the laneLink.

Let me take one example:
the normal Junction which works in Carla should looks like this:
`<junction id="295" name="junction295">
        <connection id="0" incomingRoad="70" connectingRoad="296" contactPoint="start">
            <laneLink from="-1" to="-1"/>
        </connection>
        <userData code="vectorJunction">
            <vectorJunction junctionId="{f322b59c-7c41-462d-87a0-f37793bd33fb}"/>
        </userData>
    </junction>`

the junction which throw the error looks like this:
`<junction id="299" name="junction299">
        <userData code="vectorJunction">
            <vectorJunction junctionId="{ccfcd7df-45d3-495c-b4c4-b56ee4de8468}"/>
        </userData>
    </junction>`

I hope this can helps to find the Issue a bit.

closed 2026-11-12 by glopezdiest