carla-simulator/carla #4893: The car goes over the pavement when turning

iohnishijima, 2021-11-16:
Hi, I have a problem now. I hope someone can help me!
When I'm running the automatic_control.py program, the car goes over the pavement when turning as shown in the video.
The problem occur at junctions and on turning roads.
I made map by using RoadRunner, but The waypoints are created without problems.
When running generate_traffic.py, that doesn't happen, as shown in the video.
If anyone has an idea to solve my problem, I would appreciate it if you could let me know!
thanks.

https://user-images.githubusercontent.com/93071901/142082518-0c2e8f22-950a-4edf-9f37-6d089e004673.mov

https://user-images.githubusercontent.com/93071901/142082544-85cca67e-8072-4416-9001-cf3eb66a1e73.mov

https://user-images.githubusercontent.com/93071901/142082553-1fd57031-7f76-4447-8ad6-943cb4827e4a.mov

CARLA version: 0.9.12
Platform/OS: Ubuntu18.04

jackbart94, 2021-11-17:
Hi! The main issue is that automatic_control.py doesn't run the Traffic Manager. You're probably running the behavior agent. If you want to still keep doing that, you will probably need to change the PID values for that agent. Otherwise, if you are interested in the pygame interface, you can use manual_control.py and press P to activate the Traffic Manager.

Let me know if this solves your issue!

iohnishijima, 2021-11-17:
Thank you for replying me.
What I want to do is to drive the car along WayPoints and get the control values of the car (e.g. steering, throttle), RGB images and Seg images.
I implemented it by editing Automatic_Control.py. But the problem is that cars drive on the pavement.
You said it was because I am not using Traffic_Manager.
Looking at the [code](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/automatic_control.py#:~:text=traffic_manager%20%3D%20client.get_trafficmanager()), I think it's running using Traffic_manager, is there any way to enable this if it's not?

thanks.

Phd-Ma, 2021-11-18:
> Thank you for replying me. What I want to do is to drive the car along WayPoints and get the control values of the car (e.g. steering, throttle), RGB images and Seg images. I implemented it by [...]

well done work! but i want to know how do you creat the new road? what's the soft ware you used?

jackbart94, 2021-11-18:
> Thank you for replying me. What I want to do is to drive the car along WayPoints and get the control values of the car (e.g. steering, throttle), RGB images and Seg images. I implemented it by [...]

Why don't you do this in the ``while True`` loop in generate_traffic.py? You just get the control values and generate images from there. There's no need to use ``automatic_control.py`` for that. If you need the pygame interface, just use ``manual_control.py`` as I specified in the last comment.

Also, ``automatic_control.py`` doesn't use Traffic Manager. It only creates the client, but if you don't use ``set_autopilot`` on the vehicle that has been created, it will use another agent. To use Traffic Manager then, you'd have to change the script to enable Traffic Manager and not use any of the other agents.

iohnishijima, 2021-11-19:
@Phd-Ma 
> well done work! but i want to know how do you creat the new road? what's the soft ware you used?

Hi! Thank you for replying me.
I am using RoadRunner to create new road. It is MatLab software. I highly recommend it because you can create a new map like a game.
If you are interested in this, please refer to the URL below.

https://carla.readthedocs.io/en/0.9.12/tuto_M_generate_map/

https://carla.readthedocs.io/en/0.9.12/tuto_M_add_map_source/

Thanks

iohnishijima, 2021-11-19:
> Why don't you do this in the `while True` loop in generate_traffic.py? You just get the control values and generate images from there. There's no need to use `automatic_control.py` for that. If you [...]

Thank you for replying me.
I didn't have that idea. Your reply is very good for me. I'll try it with traffic_manager.py.
I really appreciate your teaching.

Phd-Ma, 2021-11-22:
@iohnishijima 

Thanks you very much!

I have install RoadRunner software used to Create custom maps， I am try to export "Carla (*.fbx, *,osdx)" .  and then make launch for UE4editor, import the named.fbx to /content/carla/Maps, the scene had succeed import in the UE4.

While , when i click "Play" button, and perform the command "python generation_traffic.py or manucontroll.py" thow out error:

**_ERROR: unable to parse the OpenDRIVE XML string 
RuntimeError: failed to generate map
The server could not send the OpenDRIVE (.xodr) file:
Make sure it exists, has the same name of your town, and is correct._**

Have you encountered such a problem, how to solve it？

iohnishijima, 2021-11-22:
@Phd-Ma 
I didn't have the same problem in my case.
Please try again the way I did. I'll show you the steps below.

1. delete all files and folder in carla/import (without ReadMe)
2. delete "map_package" folder in carla/Unreal/CarlaUE4/Content
3. do "make import" on home/carla terminal.
4. when "make import" done, do "make launch" to open UE4Editor.
5. when you open UE4Editor, you should see some words like "built mesh[3802...]" in the bottom right corner.
6. Press Play when finished building mesh.

i hope these steps will be useful to you!
thanks.

closed 2021-11-30 by MattRowe18