carla-simulator/carla #4899: Support for ALKS Test Scenarios using OpenSCENARIO 1.1

jdsika, 2021-11-18:
Dear team,

at BMW we have published an interpretation and implementation of the [ALKS Regulation No. 157](https://github.com/asam-oss/OSC-ALKS-scenarios) using the [ASAM OpenX](https://www.asam.net/standards/detail/openscenario/). It is a community effort for a common understanding of the necessary test scenarios to be conducted and the requirements arsing from the continously developing law towards a scenario language. We are aming to showcase the usage of these test sceanrios with all available open source simulators and are most succesfull using the tools esmini from Volvo and OpenPass from BMW.

The goal is to tackle some of the questions below which in our experience are often a problem in collaborations:

- What are the right scenarios?
- Do we have the same understanding of the underlying standard?
- Do the involved tools support the used feature set of the standard?
- What general quality measures have been taken?
- How are quality and requirements connected to the use case?

With the attached instruction we provide in the README we can execute a subset of the sceanrios but also only using the OpenSCENARIO version 1.0. This excludes e.g. constraints and parameter variations which are quite important. It would be great if you would consider adding the support for these scenarios in CARLA and maybe add them to your CI build pipeline as emsini has [showcased already](https://github.com/esmini/esmini/issues/45). Please also have a note about the open PR mentioned below which never made it into the master. [Controller activation](https://github.com/asam-oss/OSC-ALKS-scenarios/pull/50) maybe an interseting toipic as well.

I would be happy to work together on this topic with my colleagues and assist you if you need anything.

Best regards
Carlo van Driesten

-------------------------------------
The execution in the open source simulator "CARLA" under Ubuntu 20.04 is described here:

1. Clone or download this repository to your local drive.
2. Download CARLA and the compatible scenario-runner for CARLA (tested successfully with [CARLA 0.9.11](https://github.com/carla-simulator/carla/releases/tag/0.9.11) and [scenario_runner-0.9.11](https://github.com/carla-simulator/scenario_runner/releases/tag/v0.9.11)
3. Follow the installation instructions for [CARLA](https://carla.readthedocs.io/en/0.9.11/) and the [scenario_runner](https://github.com/carla-simulator/scenario_runner/blob/master/Docs/getting_scenariorunner.md) (be sure to install all the required tools and libs from requirements.txt (mentioned in "Installation summary" and to set the environment variables (mentioned in "B. Download ScenarioRunner from source"))
4. Once you can run the .xosc scenarios delivered with CARLA, run the ALKS scenarios like this:
 a) Start the CARLA simulator: Go to the CARLA installation folder and type "./CarlaUE4.sh" 
 b) Start the scenario runner: Go to the scenario_runner installation folder and type e.g. "python scenario_runner.py --openscenario /path/to/OSC-ALKS-scenarios/Scenarios/ALKS_Scenario_4.1_1_FreeDriving_TEMPLATE.xosc"

_Note:_ For execution with CARLA please use the [release v0.3.2](https://github.com/asam-oss/OSC-ALKS-scenarios/releases/tag/v0.3.2) with scenarios in OpenSCENARIO 1.0 format.

Execution with esmini or openPASS on Linux hasn't been tested yet.

#### Notes regarding CARLA

With CARLA version 0.9.11 the following scenarios are supported: 4.1_1, 4.2_1, 4.2_2, 4.2_4.

_Note:_ 4.2_X scenarios do only work if the pedestrian is modeled directly in the scenario and not in a catalog or is exchanged by a vehicle. Please refer to the bug fix in [this PR](https://github.com/carla-simulator/scenario_runner/pull/713)

jdsika, 2021-12-15:
It would be great to keep a continous discussion here. Just ping if you want to setup a meeting next year if you have specific questions!

jdsika, 2022-04-18:
This should not be stale...

Palpendiculal, 2023-05-26:
Hi,
Are there any updates on the issues. I am facing similar issue of incompatible against Openscenario  Version 1.1.  As of now, Carla supports only version 1.0. 
To provide more context, I am trying to run scenarios from Safety Pool which are in ASAM Openscenario format 1.1

Thank you