carla-simulator/carla #4901: Add a command line option to start simulator with less complex map

hh0rva1h, 2021-11-18:
Starting with Carla 0.9.12, a new default Town is used that takes considerably longer to load and there seems to be no option to launch Carla with another less demanding Map. Technically there is the option to modify `CarlaUE4/Config/DefaultEngine.ini` however that's not very end-user-friendly and also often difficult to do when using the Docker image on an HPC cluster where you do not have the rights to modify containers.

To us it appears as a waste of resources to let Carla startup with that big map when we load our preferred map afterwards anyway. When computing resources are limited, e.g. when you use a laptop for prototyping, you might easily run out of VRAM with Carla loading that big map per default, while simpler maps would work just fine.

So please consider adding a command line option for selecting the preferred map, from the perspective of RAM this is as important as the `-quality-level=Low` option.

hh0rva1h, 2022-04-29:
I think it is expected from feature requests to be stale, so bumping this in order for stale bot not to close this.

krtt, 2023-05-17:
Any update here? I was just looking for this feature too and learned here that there is no such option yet from command line.