carla-simulator/carla #4905: failed make launch to CARLA 0.9.13 #4904

Isak-toast, 2021-11-19:
CARLA version: 0.9.13
Platform/OS: Ubuntu 18.04
Problem you have experienced: Print error while make launch
What you expected to happen: make launch
Other information (documentation you consulted, workarounds you tried): 
Hello!
I knew that update new version CARLA on the current week.
So I was updating it yesterday, but it's not working. 

[image]

Please give me an advice. Thank you.

afterimagex, 2021-11-19:
i have the seem error.

MaXiaoSIOM, 2021-11-19:
the same error

Axel1092, 2021-11-19:
Hi all, for the new 0.9.13 version it is necessary to update the engine code. Otherwise you will find these compilation errors.

Isak-toast, 2021-11-19:
What if can i reinstall Unreal Engine 4.26? Will it be clear?

I don't know exactly what you mean engine code..

Isak-toast, 2021-11-22:
Yeah! It's work. 
I have reinstalled Unreal Engine 4.26.2

CARLA Version : 0.9.13 

Thanks.

closed 2021-11-22 by Isak-toast

huangrainqi, 2021-11-24:
the same error

huangrainqi, 2021-11-24:
@Isak-toast Where to download  Unreal Engine 4.26.2 ?  the lastest version is 4.26.1 ?