carla-simulator/carla #4906: Why is carla/Build/patchelf-install/ empty?

MR-hyj, 2021-11-19:
When I was running `make PythonAPI`, the error occurs like
> cp: unable to reach /my/path/to/carla/Build/patchelf-install/bin/patchelf: no such file or directory

So I take a look at the `carla/Build/patchelf-install/` directory, and find it is totally empty.
Luckily, I find `carla/Build/patchelf-src/`, and do the following just as `carla/Build/patchelf-src/README.md` says
> ./bootstrap.sh
> ./configure
> make
> make check
> sudo make install

It all worked, but still the `carla/Build/patchelf-install/` is empty.
And when I was running the above 5 cmds, I found no information indicating they are targeted to `carla/Build/patchelf-install/`. It seems it is installed somewhere else.

MR-hyj, 2021-11-19:
I've solved this problem.
`whereis patchelf` shows the patchelf is installed at another directory. So I created an link to `carla/Build/patchelf-install/bin/parchelf`. 
God damn it works!

MattRowe18, 2021-11-30:
@MR-hyj thanks for posting this solution, I will close this issue now

closed 2021-11-30 by MattRowe18

laurenceandrews, 2022-11-06:
> I've solved this problem. `whereis patchelf` shows the patchelf is installed at another directory. So I created a link to `carla/Build/patchelf-install/bin/parchelf`. God damn it works!

Hey, would you mind letting me know where you found the goods? Was it the patchelf-src folder? And how did you create the link?

cocobear1223, 2022-11-07:
plz,specify your solution steps.THX

tafer96, 2023-06-03:
1. Install patchelf:
_sudo apt-get update
sudo apt-get install patchelf_
2. check where patchelf is located:
_whereis patchelf_
3. copy the file in the location, create a bin folder in patchelf-install, place the file in the bin folder.

Hope this description is helping you guys!