carla-simulator/carla #4907: make launch raise 18+10 errors

MR-hyj, 2021-11-19:
multiple errors say "use of dynamic_cast requires -frtti" (Where can I find the g++ cmds so I can add the `-frtti` argument?)
other errors range from "no member named xxx", to "base class xxx has no member xxx". (May be the head files are mis-included?)

> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.h:82:56: error: base class has incomplete type
> class FTaggedHierarchicalStaticMeshSceneProxy : public FHierarchicalStaticMeshSceneProxy
>                                                 ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.h:82:7: note: definition of 'FTaggedHierarchicalStaticMeshSceneProxy' is not complete until the closing '}'
> class FTaggedHierarchicalStaticMeshSceneProxy : public FHierarchicalStaticMeshSceneProxy
>       ^
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.h:87:35: error: 'GetViewRelevance' marked 'override' but does not override any member functions
>   virtual FPrimitiveViewRelevance GetViewRelevance(const FSceneView * View) const override;
>                                   ^
> In file included from /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.1_of_3.cpp:23:
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:170:10: error: cannot initialize return object of type 'FPrimitiveSceneProxy *' with an rvalue of type 'FTaggedHierarchicalStaticMeshSceneProxy *'
>                 return new FTaggedHierarchicalStaticMeshSceneProxy(MeshComponent, bIsGrass, GetWorld()->FeatureLevel, TaggedMID);
>                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:237:97: error: no member named 'NotDrawTaggedComponents' in 'FEngineShowFlags'
>   ViewRelevance.bDrawRelevance = ViewRelevance.bDrawRelevance && !View->Family->EngineShowFlags.NotDrawTaggedComponents;
>                                                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:265:97: error: no member named 'NotDrawTaggedComponents' in 'FEngineShowFlags'
>   ViewRelevance.bDrawRelevance = ViewRelevance.bDrawRelevance && !View->Family->EngineShowFlags.NotDrawTaggedComponents;
>                                                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:291:97: error: no member named 'NotDrawTaggedComponents' in 'FEngineShowFlags'
>   ViewRelevance.bDrawRelevance = ViewRelevance.bDrawRelevance && !View->Family->EngineShowFlags.NotDrawTaggedComponents;
>                                                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:300:5: error: member initializer 'FHierarchicalStaticMeshSceneProxy' does not name a non-static data member or base class
>   : FHierarchicalStaticMeshSceneProxy(bInIsGrass, Component, InFeatureLevel)
>     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:305:3: error: use of undeclared identifier 'bVerifyUsedMaterials'
>   bVerifyUsedMaterials = false;
>   ^
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:307:8: error: unknown type name 'FLODInfo'
>   for (FLODInfo& LODInfo : LODs) {
>        ^
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:307:28: error: use of undeclared identifier 'LODs'
>   for (FLODInfo& LODInfo : LODs) {
>                            ^
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:308:10: error: use of undeclared identifier 'FLODInfo'
>     for (FLODInfo::FSectionInfo& SectionInfo : LODInfo.Sections) {
>          ^
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:316:43: error: use of undeclared identifier 'FHierarchicalStaticMeshSceneProxy'; did you mean 'FTaggedHierarchicalStaticMeshSceneProxy'?
>   FPrimitiveViewRelevance ViewRelevance = FHierarchicalStaticMeshSceneProxy::GetViewRelevance(View);
>                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
>                                           FTaggedHierarchicalStaticMeshSceneProxy
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.h:82:7: note: 'FTaggedHierarchicalStaticMeshSceneProxy' declared here
> class FTaggedHierarchicalStaticMeshSceneProxy : public FHierarchicalStaticMeshSceneProxy
>       ^
> In file included from /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.1_of_3.cpp:23:
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:318:97: error: no member named 'NotDrawTaggedComponents' in 'FEngineShowFlags'
>   ViewRelevance.bDrawRelevance = ViewRelevance.bDrawRelevance && !View->Family->EngineShowFlags.NotDrawTaggedComponents;
>                                                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
> 30 warnings and 4 errors generated.
> In file included from /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.1_of_3.cpp:1:
> In file included from /media/user/BA72C98372C9453B/UnrealEngine_4.26/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/UnrealEd/SharedPCH.UnrealEd.h:280:
> In file included from /media/user/BA72C98372C9453B/UnrealEngine_4.26/Engine/Source/Editor/UnrealEd/Public/UnrealEdSharedPCH.h:269:
> /media/user/BA72C98372C9453B/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Public/Templates/Casts.h:442:10: error: use of dynamic_cast requires -frtti
>                 return dynamic_cast<To>(Arg);
>                        ^
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/CarlaActor.h:169:12: note: in instantiation of function template specialization 'UE4Casts_Private::DynamicCast<FVehicleData *, FActorData>' requested here
>     return dynamic_cast<T*>(ActorData.Get());
>            ^
> /media/user/BA72C98372C9453B/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Public/Templates/Casts.h:469:40: note: expanded from macro 'dynamic_cast'
> #define dynamic_cast UE4Casts_Private::DynamicCast
>                                        ^
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/CarlaActor.cpp:636:31: note: in instantiation of function template specialization 'FCarlaActor::GetActorData<FVehicleData>' requested here
>     FVehicleData* ActorData = GetActorData<FVehicleData>();
>                               ^
> In file included from /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.1_of_3.cpp:1:
> In file included from /media/user/BA72C98372C9453B/UnrealEngine_4.26/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/UnrealEd/SharedPCH.UnrealEd.h:280:
> In file included from /media/user/BA72C98372C9453B/UnrealEngine_4.26/Engine/Source/Editor/UnrealEd/Public/UnrealEdSharedPCH.h:269:
[... 77 lines cut ...]
>                        ^
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/CarlaActor.h:175:12: note: in instantiation of function template specialization 'UE4Casts_Private::DynamicCast<FTrafficLightData *, FActorData>' requested here
>     return dynamic_cast<T*>(ActorData.Get());
>            ^
> /media/user/BA72C98372C9453B/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Public/Templates/Casts.h:469:40: note: expanded from macro 'dynamic_cast'
> #define dynamic_cast UE4Casts_Private::DynamicCast
>                                        ^
[... cut ...]
> /media/user/BA72C98372C9453B/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Public/Templates/Casts.h:469:40: note: expanded from macro 'dynamic_cast'
> #define dynamic_cast UE4Casts_Private::DynamicCast
>                                        ^
> /media/user/BA72C98372C9453B/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/WorldObserver.cpp:225:46: note: in instantiation of function template specialization 'FCarlaActor::GetActorData<FTrafficSignData>' requested here
>     const FTrafficSignData* ActorData = View.GetActorData<FTrafficSignData>();
>

bernatx, 2021-11-19:
Did you upgraded the UE4 engine? 0.9.13 needs changes in the unreal engine

huhaocong1, 2021-11-30:
i  met the same problem,how did you solve that?

Yandong-Luo, 2021-12-02:
I met the same problem too. And my UE4 engine is 4.26 and the carla is 0.9.13

huhaocong, 2021-12-02:
> 

> I met the same problem too. And my UE4 engine is 4.26 and the carla is 0.9.13

I fixed the problem:because the UE4 version i used was 4.26.1,must use 4.26.2!
So ,you can update the UE4's vetsion to solve the problem either.

Yandong-Luo, 2021-12-02:
> 

I could not find the 4.26.2 from branch. I just only can found 4.26 and 4.26-chaos. So 4.26-chaos is the 4.26.2?

Yandong-Luo, 2021-12-02:
> > 
> 
> > I met the same problem too. And my UE4 engine is 4.26 and the carla is 0.9.13
> 
> I fixed the problem:because the UE4 version i used was 4.26.1,must use 4.26.2! So ,you can update the UE4's version to solve the problem either.

So do you use the carla 0.9.13? When I have used 4.26.2, the error stills the same as before.

bernatx, 2021-12-02:
Hi,
Just to clarify, the UE4 that you need to use with CARLA 0.9.13 is this one (UE4 fork): https://github.com/CarlaUnreal/UnrealEngine
Regards

xiaose26, 2021-12-11:
have you solved these errors? 
my version:
- carlar: "0.9.13-1-g8854804f"
- UE: 4.26.2

Hardliu, 2021-12-23:
Hi, I met the same problem, who can solve it ?
CARLA 0.9.12
UE4 4.26.2

den1den, 2022-01-20:
Ensure you have the correct UE4_ROOT variable set. For me the command `set UE4_ROOT=F:\unreal` fixed this problem. I have checkout out (and build) the [CARLA fork of the Unreal Engine](https://github.com/CarlaUnreal/UnrealEngine) to F:\unreal

Rocky0102, 2022-02-10:
Please check this: https://github.com/CarlaUnreal/UnrealEngine/pull/2

MR-hyj, 2022-02-10:
I figured this out by turning to `release` version ( [here)](https://github.com/carla-simulator/carla/releases) instead of rebuilding from source.

Anirudh58, 2022-04-05:
I am building unreal 4.26.2 through ue4-build as: 

`ue4-docker build --cuda=10.2 --no-minimal --no-full --exclude=debug --exclude=templates 4.26.2`

And then building carla 0.9.13 as: 

```
# Build Carla
RUN git clone https://github.com/carla-simulator/carla $CARLA_PATH && \
    cd $CARLA_PATH && \
    git checkout $CARLA_VERSION && \
    ./Update.sh && \
    make package
```

I get the same error. Any tips on how to build successfully?