carla-simulator/carla #4926: Relative position of Target wrt Ego Vehicle

werewolfdev, 2021-11-24:
Hi Team, Iam trying to log both ego and target vehicle position and get the relative position of target vehicle wrt Ego. I have kept pitch and roll as 0 and just yaw is present . SO its like Iam playing in 2D. I would like to confirm if my methodology is correct or not .

Initially I determine the 2   3*3 homogeneous matrices T_ego and T_target which describes the translation and rotation of the actor, w.r.t Unreal world frame.
The relative pose of the target w.r.t ego-vehicles is given by T_rel = inv(T_ego) * T_targ.
The relative position is T_rel[:2, 2]
The relative rotation (from target axes to ego-vehicle axes) is T_rel[:2, :2].

CARLA version:0.9.11
Platform/OS:Ubuntu 18.04

Thanks a lot in advance.

Chris1nexus, 2021-11-25:
I am not from the carla simulator Team. However i have been working on simulation of vision and trajectories, expressed in relative pose of the ego vehicle.
Your approach is correct, as your usage utilizes the traditional conventions of roto-translations in homog coords.
Nevertheless, you should try to plot on 2D-plane both the position vector of ego and target vehicles, just to be sure that you are treating the matrices correctly also in practice and not only in theory.

werewolfdev, 2021-11-25:
Hi @Chris1nexus  thanks a lot for the reply and for confirming the approach. Yes I am carrying out some unit tests and visualization based tests of output of the script developed and it seems promising.

closed 2021-11-30 by MattRowe18