carla-simulator/carla #4927: Retrieving high-level commands for autopilot hero vehichle

zachrenlee, 2021-11-24:
Hi,
is there any way to obtain the high-level commands (LEFT, RIGHT, STRAIGHT, LANEFOLLOW, CHANGELANELEFT, CHANGELANERIGHT) from the autopilot agent? I am collecting data using the autopilot mode by setting the hero_actor.set_autopilot(True).
Thank you very much.

jackbart94, 2021-11-25:
Hi, we just introduced this feature. It's enough to do ``traffic_manager.get_next_action(hero_actor)`` to get the next high level action, and ``traffic_manager.get_all_actions(hero_actor)`` to get all the high-level choice that the autopilot has already computed in its path buffer.

It's a very straightforward answer so I'll close this issue, let me know if it works for you though.

closed 2021-11-25 by jackbart94

zachren97, 2021-11-25:
> Hi, we just introduced this feature. It's enough to do `traffic_manager.get_next_action(hero_actor)` to get the next high level action, and `traffic_manager.get_all_actions(hero_actor)` to get all [...]

Hi, may I know which version of Carla has this feature? I tried it with 0.9.10 and apparently 'TrafficManager' object has no attribute 'get_next_action'. Also is it possible to pass a high level command to the autopilot algorithm?