carla-simulator/carla #4928: spectator perspective problem

whuxhzhang, 2021-11-24:
Ubuntu18.04  Carla-0.9.13

Set the spectator to be related to the ego vehicle, 
and the spectator perspective will swing greatly as the vehicle accelerates and decelerates.

code is:
spectator.set_transform(carla.Transform(ego_trans.location + carla.Location(x=0, y=0, z=1.3),
                                                                carla.Rotation(yaw=ego_yaw)))

MattRowe18, 2021-11-30:
@whuxhzhang hi

could you provide us a little bit more information about what happens, it's not clear from your code how the problem is arising. Does the spectator swing from side to side? A couple of screenshots or a video would help a lot in debugging this issue

thanks

whuxhzhang, 2021-12-01:
[image]
[image]

These two pictures are screenshots of the same car's perspective when accelerating and decelerating

code is:
spectator.set_transform(carla.Transform(ego_trans.location + carla.Location(x=0, y=0, z=1.3),
carla.Rotation(yaw=ego_yaw)))

closed 2022-05-02 by stale[bot]

flyico, 2022-09-01:
I have same issue, have you resolved this?

Jasmineyangyangyang, 2023-04-26:
I have same problem, have you resolved this?

mamaduke21, 2023-05-31:
Same problem here and it is worst when using VR, help?