carla-simulator/carla #4929: Error when I run the command "make Python API"

umlautsim, 2021-11-24:
Hello, I tried to compile the pyhton API client using the command "make python API" and I got this error. Using ubuntu 18.04.

Snippet of the error:
[](url)
1 error generated.

  "clang++" -c -x c++ -fvisibility-inlines-hidden -m64 -pthread -O3 -Wall -fvisibility=hidden -Wno-inline -fPIC -std=c++14 -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_ALL_NO_LIB=1 -DBOOST_PYTHON_SOURCE -DBOOST_PYTHON_STATIC_LIB -DNDEBUG -I"." -I"/usr/include/python3.7" -o "bin.v2/libs/python/build/clang-linux-8.0/release/link-static/python-3.7/threading-multi/visibility-hidden/object/function_doc_signature.o" "libs/python/src/object/function_doc_signature.cpp"

...failed clang-linux.compile.c++.without-pch bin.v2/libs/python/build/clang-linux-8.0/release/link-static/python-3.7/threading-multi/visibility-hidden/object/function_doc_signature.o...
...skipped <pbin.v2/libs/python/build/clang-linux-8.0/release/link-static/python-3.7/threading-multi/visibility-hidden>libboost_python37.a(clean) for lack of <pbin.v2/libs/python/build/clang-linux-8.0/release/link-static/python-3.7/threading-multi/visibility-hidden>list.o...
...skipped <pbin.v2/libs/python/build/clang-linux-8.0/release/link-static/python-3.7/threading-multi/visibility-hidden>libboost_python37.a for lack of <pbin.v2/libs/python/build/clang-linux-8.0/release/link-static/python-3.7/threading-multi/visibility-hidden>list.o...
...skipped <p/home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib>libboost_python37.a for lack of <pbin.v2/libs/python/build/clang-linux-8.0/release/link-static/python-3.7/threading-multi/visibility-hidden>libboost_python37.a...
...skipped <pbin.v2/libs/python/build/clang-linux-8.0/release/link-static/python-3.7/threading-multi/visibility-hidden>libboost_python-variant-static-py3.7.cmake for lack of <pbin.v2/libs/python/build/clang-linux-8.0/release/link-static/python-3.7/threading-multi/visibility-hidden>libboost_python37.a...
clang-linux.compile.c++.without-pch bin.v2/libs/program_options/build/clang-linux-8.0/release/link-static/threading-multi/visibility-hidden/utf8_codecvt_facet.o
clang-linux.compile.c++.without-pch bin.v2/libs/program_options/build/clang-linux-8.0/release/link-static/threading-multi/visibility-hidden/config_file.o
clang-linux.compile.c++.without-pch bin.v2/libs/program_options/build/clang-linux-8.0/release/link-static/threading-multi/visibility-hidden/convert.o
boost-install.generate-cmake-config- bin.v2/libs/filesystem/build/stage/boost_filesystem-config.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_filesystem-1.72.0/boost_filesystem-config.cmake
boost-install.generate-cmake-config-version- bin.v2/libs/filesystem/build/stage/boost_filesystem-config-version.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_filesystem-1.72.0/boost_filesystem-config-version.cmake
...on 200th target...
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/Boost-1.72.0/BoostConfig.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/Boost-1.72.0/BoostConfigVersion.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_headers-1.72.0/boost_headers-config.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_headers-1.72.0/boost_headers-config-version.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_program_options-1.72.0/boost_program_options-config.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_program_options-1.72.0/boost_program_options-config-version.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_program_options-1.72.0/libboost_program_options-variant-shared.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_python-1.72.0/boost_python-config.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_python-1.72.0/boost_python-config-version.cmake
...skipped <p/home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_python-1.72.0>libboost_python-variant-shared-py3.7.cmake for lack of <pbin.v2/libs/python/build/clang-linux-8.0/release/python-3.7/threading-multi/visibility-hidden>libboost_python-variant-shared-py3.7.cmake...
...skipped <p/home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_python-1.72.0>libboost_python-variant-static-py3.7.cmake for lack of <pbin.v2/libs/python/build/clang-linux-8.0/release/link-static/python-3.7/threading-multi/visibility-hidden>libboost_python-variant-static-py3.7.cmake...
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_numpy-1.72.0/boost_numpy-config.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_numpy-1.72.0/boost_numpy-config-version.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_system-1.72.0/boost_system-config.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_system-1.72.0/boost_system-config-version.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_system-1.72.0/libboost_system-variant-shared.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_system-1.72.0/libboost_system-variant-static.cmake
clang-linux.compile.c++.without-pch bin.v2/libs/program_options/build/clang-linux-8.0/release/link-static/threading-multi/visibility-hidden/variables_map.o
clang-linux.compile.c++.without-pch bin.v2/libs/program_options/build/clang-linux-8.0/release/link-static/threading-multi/visibility-hidden/cmdline.o
clang-linux.compile.c++.without-pch bin.v2/libs/program_options/build/clang-linux-8.0/release/link-static/threading-multi/visibility-hidden/winmain.o
clang-linux.compile.c++.without-pch bin.v2/libs/program_options/build/clang-linux-8.0/release/link-static/threading-multi/visibility-hidden/options_description.o
clang-linux.compile.c++.without-pch bin.v2/libs/program_options/build/clang-linux-8.0/release/link-static/threading-multi/visibility-hidden/parsers.o
clang-linux.compile.c++.without-pch bin.v2/libs/program_options/build/clang-linux-8.0/release/link-static/threading-multi/visibility-hidden/value_semantic.o
clang-linux.compile.c++.without-pch bin.v2/libs/program_options/build/clang-linux-8.0/release/link-static/threading-multi/visibility-hidden/split.o
clang-linux.archive bin.v2/libs/program_options/build/clang-linux-8.0/release/link-static/threading-multi/visibility-hidden/libboost_program_options.a
boost-install.generate-cmake-variant- bin.v2/libs/program_options/build/clang-linux-8.0/release/link-static/threading-multi/visibility-hidden/libboost_program_options-variant-static.cmake
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/libboost_program_options.a
common.copy /home/umlaut01/Carla/carla-master/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_program_options-1.72.0/libboost_program_options-variant-static.cmake
...failed updating 54 targets...
...skipped 12 targets...
...updated 171 targets...
Util/BuildTools/Linux.mk:137: recipe for target 'setup' failed
make: *** [setup] Error 1

Kzmc-China, 2021-11-27:
I have the same error. Have you found a solution to this?

EatAllBugs, 2021-12-01:
how do you fix it?

Dou-Yuxiao, 2021-12-03:
I got the same error

umlautsim, 2021-12-06:
Hi, I fixed this error by installing python Dev. Here is the command I ran to fix the issue
`apt-get install python-dev`

fatemehjdi, 2022-11-09:
I have the same issue. Can anyone help me with it?

leishi23, 2022-11-14:
> 

I tried `sudo apt-get install python-dev` and then make PythonAPI again, but failed as the same.SAD :(

Yandong-Luo, 2023-01-29:
I think you can change your python version to compile it. python 3.9 and python 3.6 works for me, but python 3.7 could not work.

mohussein97, 2024-05-05:
I have same problem Python 3.8 carla .9.15 ubuntu 20

darshandodamani, 2024-06-23:
I have the same issue with python 3.10

Jenson66, 2024-10-10:
I have the same issue with python 3.10

leishi23, 2024-10-10:
> > 
> 
> I tried `sudo apt-get install python-dev` and then make PythonAPI again, but failed as the same.SAD :(

Hi guys~ I forgot to update this. 

1. Ensure the CARLA is build from Linux/Windows instead of quick package installation.
2. The clang c++ package always has weird bugs... I have met more than three different issues on this. Stackoverflow always helps me fix it.

CARLA is easy to start, but needs some tricks when diving into. Good luck!