carla-simulator/carla #4932: Repository to build Carla on windows doesn't exist

RocaPiedra, 2021-11-28:
Hi, 

I started using Carla for my master's thesis and recently I've purchased a laptop (Gigabyte aero15xd) because my old PC couldn't handle Carla properly. So far, i've not been able to make the nvidia drivers work on ubuntu so I decided to build Carla on windows even though I prefer linux for programming. I've tried to build several versions of Carla but the unreal engine repository (https://github.com/CarlaUnreal/UnrealEngine.git/) doesn't exist:

[image]

Do you know how can I solve this? Should I go for previous versions as 0.9.11?
I really need to start working with it asap, so if you have built Carla in windows recently or know how to solve the nvidia drivers issue properly (I've reinstalled ubuntu 3 times already), please answer.

MattRowe18, 2021-11-30:
@RocaPiedra hi

Have you created an account on Epic Games, you need to connect your Github account with Epic Games and get a token for authentication before you can download the CARLA UE4 fork

RocaPiedra, 2021-12-06:
Hi Matt, sorry I missed your answer.
I did it (checked with my epic account and it is linked), and git clonned with the x64 cmd too. I managed to use Carla with the precompiled .tar.gz folder.
The thing is, I have no clue which Unreal Engine is working as I have 4.24 and 4.26 versions installed nad not really know how I can access the uproject. I simply launch the exe file in the precompiled folder and I 'm ready to launch clients.

hgc1, 2022-02-09:
@RocaPiedra hi，
I was checked with my epic account and it is linked, but it still no found. Can you elaborate on how you solved the problem？ Please answer.
[image]
[image]

closed 2022-02-15 by RocaPiedra

FrenZ0810, 2024-01-31:
Hi RocaPiedra
I have encountered the same problem as you.May I ask how you specifically resolved it?I used two methods suggested to connect Github account and Epic Games, but it still doesn't work.
[image]
[image]
[image]
[image]

Emilyhelin122, 2024-06-24:
I had the same problem with you. Do you solve the problem?