carla-simulator/carla #4933: After installing ROS noetic make package/import do not work/

rrrushan, 2021-11-28:
Hello!
CARLA version: 0.9.12 and 0.9.13
Platform/OS: Ubuntu 20.04

After installing ROS noetic i get the next error: 

building 'carla.libcarla' extension
clang-8 -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Idependencies/include -I/usr/include/python3.8 -c source/libcarla/libcarla.cpp -o build/temp.linux-x86_64-3.8/source/libcarla/libcarla.o -isystem dependencies/include/system -fPIC -std=c++14 -Werror -Wall -Wextra -Wpedantic -Wno-self-assign-overloaded -Wdeprecated -Wno-shadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true

In file included from source/libcarla/libcarla.cpp:7:
In file included from dependencies/include/carla/Memory.h:9:
In file included from dependencies/include/system/boost/enable_shared_from_this.hpp:16:
In file included from dependencies/include/system/boost/smart_ptr/enable_shared_from_this.hpp:16:
dependencies/include/system/boost/smart_ptr/weak_ptr.hpp:16:10: fatal error: 
      'memory' file not found
#include <memory> // boost.TR1 include order fix
         ^~~~~~~~
1 error generated.
error: command 'clang-8' failed with exit status 1
make: *** [Util/BuildTools/Linux.mk:87: PythonAPI] Error 1

I am sure the problem is in the dependencies.
Once I removed all versions gcc and installed back, make import started to work again, but the ROS was gone.
If anyone knows where to dig, please help me with it.
Thanks!

rrrushan, 2021-11-28:
ROS noetic Desktop-Full was installed.
I had to apply to [this](https://github.com/carla-simulator/carla/issues/2792#issuecomment-898733713)
In order to make 'make PythonAPI' work again I had to enter into the terminal:
sudo apt-get purge gcc-11
sudo apt autoremove
sudo apt-get install build-essential

But I want to keep working ROS on my system.
I will try to install the minimal setup of ROS and see if it works.

closed 2022-05-02 by stale[bot]

naruarjun, 2023-01-06:
I am also facing the same issue, has it been resolved?

naruarjun, 2023-01-06:
Hello,
I am facing this same issue. Has it been resolved?

miaoxw, 2023-07-10:
This issue happens to me too. I am running ROS noetic and trying to build CARLA 0.9.14 on the same PC.

miaoxw, 2023-07-10:
FYI, perhaps we need to specify a version of libstdc++ to install.
https://github.com/carla-simulator/carla/issues/6139#issuecomment-1628046496