carla-simulator/carla #4934: Town04.xodr incomplete conversion via netconvert_carla.py

GittUser2021, 2021-11-29:
I been trying to figure out why I can’t fully convert town 4 using netconvert_carla.py. everything seems to look fine except for the road circled in the attached image. When I launch town04 in Carla, it is a gas station and it does show in the net file that is in the examples but when I try to convert it, it ignores it. 
I tired modifying the “opendrive_netconvert.typ.xml” by duplicating the line “     <type id="driving"       priority="1"  speed="13.89" width="3.65" disallow="pedestrian rail rail_urban rail_electric rail_fast tram ship"/>  “ but modifying  the width to 3.50 but still no luck. Any advice on how to proceed with that? 

[image]
[image]

Carla version: 0.9.13 and 0.9.12

-->

GittUser2021, 2021-11-30:
made some progress with the conversion, I updated to Sumo 1.11.0 and I was able to convert the maps and be able to get the side road (gas station) see picture below. 
 
[image]

however, still look different to the original curve from Carla Town04.net.xml example. But when the run the Town04.xodr file in roadrunner my side road looks similar to the one from RoadRunner. See picture below. Still looking for a solution/explanation on why the road looks curvy vs a straight road. 

[image]

joel-mb, 2021-12-01:
Hi @GittUser2021,
As far as I remember, this could be happening because the arguments used in the `netconvert` command to generate the net file in the `examples` folder were different to the ones used in the `netconvert_carla.py` script (https://github.com/carla-simulator/carla/blob/master/Co-Simulation/Sumo/util/netconvert_carla.py#L377-L389). Here you check all the parameters available for `netconvert` (https://sumo.dlr.de/docs/netconvert.html). For instance, setting the parameter `--opendrive.curve-resolution` to 0.25 reproduces the same output you obtained with SUMO 1.11.0.

GittUser2021, 2021-12-20:
Thank you for the feedback. do you have the link for the script used in the examples?

namdre, 2022-10-10:
Another option you could use is **--default.junctions.radius 0**