carla-simulator/carla #4935: Carla 0.9.13 Segmentation fault (Bad file descriptor)

hh0rva1h, 2021-11-29:
CARLA 0.9.13 (docker image)
Platform: Linux

We are observing the following segfault repeatedly in our experiments with Carla:

```
LowLevelFatalError [File:Unknown] [Line: 136]
Exception thrown: close: Bad file descriptor
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
CommonUnixCrashHandler: Signal=11
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
```
Unfortunately we do not yet have a minimal example to reproduce and dropping all of our code here is not an option, but the crash happens quite reliably: out of 10 experiments we do, at least 3 crash after some time due to this segfault.

I don't know whether the [Line: 136] gives you any hint where the problem might lie, if not, please share some instructions how we could help you with debugging this issue.

Might be related: https://github.com/carla-simulator/carla/issues/3197, https://github.com/carla-simulator/carla/issues/3994

varunjammula, 2022-01-30:
Any updates on this issue? I am having the same problem when I try to run in an episodic manner.

varunjammula, 2022-02-01:
@bernatx Any  updates on this?

aishgupta, 2022-04-03:
@bernatx Any help on this?

PatrickFSLin, 2023-03-14:
Any update? I've been stuck and tried to solve this issue for over a week, but got nothing further...even with carla 0.9.14 #5611

hjh0119, 2023-06-25:
I have the same issue , any solution?

omitheaussie, 2023-10-16:
I also have the same issue with 0.9.14, any solutions?

Leviathan321, 2023-11-19:
I am experiencing the same issue. Is this already solved?

lumurillo, 2024-01-05:
I am having the same issue even in 0.9.15. Any suggestion?

hdp-alt, 2024-07-24:
[image]

Blyron, 2024-11-06:
For those in UE4 versions please share the repro steps to get this crash. We cannot do anything only with that information. We need repro steps for crashes. 

For UE5 ( this documentation is WIP ) https://carla-ue5.readthedocs.io/en/latest/start_quickstart/ you need at least 16GB VRAM

closed 2025-04-01 by Blyron