carla-simulator/carla #4937: can't install python3.8 carla package on ubuntu 20.04 container

PinkJohnOfUs, 2021-11-30:
Problem you have experienced:
Not possible to install carla python package for python3.8 on docker container 

What you expected to happen:
installation should work with: `pip install carla`

CARLA version: 9.12
Platform/OS: Ubuntu 20.04.3 LTS -> docker container
```bash
$user@ros-bridge-container: lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.3 LTS
Release:        20.04
Codename:       focal
```
python version:
```bash
$user@ros-bridge-container:  python --version
Python 3.8.10
```

Steps to reproduce:

I do not understand why pip tries to download 0.9.5 and not 0.9.12 or 13
```
$user@ros-bridge-container: pip install carla
Collecting carla
  Using cached carla-0.9.5.tar.gz (3.9 kB)
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-ves_t_7m/carla/setup.py'"'"'; __file__='"'"'/tmp/pip-install-ves_t_7m/carla/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-ves_t_7m/carla/pip-egg-info
         cwd: /tmp/pip-install-ves_t_7m/carla/
    Complete output (7 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-ves_t_7m/carla/setup.py", line 114, in <module>
        ext_modules=get_libcarla_extensions(),
      File "/tmp/pip-install-ves_t_7m/carla/setup.py", line 31, in get_libcarla_extensions
        linux_distro = platform.dist()[0]  # pylint: disable=W1505
    AttributeError: module 'platform' has no attribute 'dist'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

Other information (documentation you consulted, workarounds you tried):
I've also downloaded wheel directly and tried:
```
pip install carla-0.9.12-cp38-cp38-manylinux_2_27_x86_64.whl 
ERROR: carla-0.9.12-cp38-cp38-manylinux_2_27_x86_64.whl is not a supported wheel on this platform.
```

joel-mb, 2021-12-01:
Hi @PinkJohnOfUs,
You require `pip` version 20.3 or above to install the CARLA Python package. You can upgrade `pip` in the following way:
```
pip install --upgrade pip
```

If you want to install the python package for CARLA 0.9.12, you should specify the version (currently, the latest version of CARLA is 0.9.13):
```
pip install carla==0.9.12
```

PinkJohnOfUs, 2021-12-01:
Thank you that solved my issue. 
For all who do have the same problem with ros:foxy-ros-base
you should do pip upgrade within dockerfile,  it wasn't possible for me to upgrade pip within running container.

closed 2021-12-01 by joel-mb

Sultan91, 2023-06-11:
```
pip install carla==0.9.12

ERROR: Could not find a version that satisfies the requirement carla==0.9.12 (from versions: 0.9.5)
ERROR: No matching distribution found for carla==0.9.12
```

Youngzipper, 2023-10-07:
> ```
> pip install carla==0.9.12
> 
> ERROR: Could not find a version that satisfies the requirement carla==0.9.12 (from versions: 0.9.5)
> ERROR: No matching distribution found for carla==0.9.12
> ```

I have the same problem. Have you got a solution yet?

mbsomeday, 2024-03-18:
I have the same version of ubuntu and python, and the error message. I solved it by this:

`sudo apt install python=3.7`
`sudo gedit ~/.bashrc`
Add this in the last line: `alias python3='/usr/bin/python3.7'`
`source ~/.bashrc`
Then try
`pip install carla`

ajay1606, 2024-10-24:
Any fixes available for installing carla with python=3.8, in ubuntu 20.4 ?

Appreciate any response.