carla-simulator/carla #4938: instance segmentation sensor documentation

guygo12345, 2021-11-30:
Hi
I'm want to use the new instance_segmentation sensor in carla 0.9.13, but there is no documentation for that. Can you please add it?
Also - are the instance ids in the image compatible with the objects ids in the world? I would like to couple the instance in the image with the instance in 3d

Thanks,

Guy

Paniac, 2021-12-07:
Hi @guygo12345,

I am also waiting for the documentation.
But as a start: the implementation is the same as with other cameras e.g. semantic segmentation. You just have to use `sensor.camera.instance_segmentation` as its blueprint. And according to the [release notes](https://carla.org/2021/11/16/release-0.9.13/) the output is encoded like this:

> Instance semantic IDs are now available embedded in the G and B channels of the RGB output of the sensor data, alongside the standard semantic IDs in the R channel.

Best,
Paniac

guygo12345, 2021-12-08:
Thanks!
I tried and managed to get the data. 
Still need to figure out how to parse the G an B together, and understand whether it's same as the 3d objects ids.

ptr-br, 2021-12-09:
I'm also very curious about the topic.

Please tell me if you find out how the G&B channels are connected to the actor id.

Schuck84, 2021-12-13:
Hi,
I'm also interested, as the environment object ID is also different than the actor ID and instance ID in the GB Pixels...

Could someone give advice?

Regards

Schuck84, 2022-01-14:
Hello,
does nobody knows or can help about this issue?
I think the problem is that the Actor ID for the Pixels are used from the Unreal actor registry and the the ID over PYTHON is catchend from the CARLA Actor Registry and there is no sync between them.

Could someone provide a fix for this?

Thanks

dbac-igd, 2022-02-04:
> Hello, does nobody knows or can help about this issue? I think the problem is that the Actor ID for the Pixels are used from the Unreal actor registry and the the ID over PYTHON is catchend from the [...]

I think you're right. If I compare the code from RayCastSemanticLidar.cpp they use the unique ID from a view they got from a registry but in the Tagger.cpp file they use the actor's unique ID directly.

pespen, 2022-02-22:
> I think you're right. If I compare the code from RayCastSemanticLidar.cpp they use the unique ID from a view they got from a registry but in the Tagger.cpp file they use the actor's unique ID [...]

Agreed. I am doing a project where I try to automatically annotate spawned objects, and this sensor would be very helpful to tighten up the 3d bounding boxes into 2d. However, since there is no link to the actor ID you get from the Carla world.get_actors() function it is not very helpful

lyfs, 2022-02-25:
[image]
I have the same question with this, if we can't get the actor's id, we have to get every object's bounding box to match it

lyfs, 2022-04-28:
您好！我是飞哥的邮箱小飞，谢谢您的来信，他会尽快回复的，祝您快乐每一天！！

HankKung, 2022-06-21:
hi 
has someone figured out the way to connect the actor id and the instance id?

Kin-Zhang, 2023-02-13:
Some discussion: [I figured out how to get Instance Segmenation color of vehicles](https://github.com/carla-simulator/carla/discussions/5047)

Other related issues: [Instance Segmentation - Standard semantic ID only contains subset of objects](https://github.com/carla-simulator/carla/issues/5842)

qhaas, 2024-01-23:
Without the ability to map actor IDs to instance semantic IDs the value of this sensor is diminished and it is, imo, counter-intuitive that they are not the same.

Blyron, 2024-11-06:
Due to a contribution from @AreopagX Now the actor id generated by carla matches the instance segmentation one; 
For docs check; https://carla.readthedocs.io/en/latest/ref_sensors/#instance-segmentation-camera 
https://carla.readthedocs.io/en/latest/tuto_G_instance_segmentation_sensor/
https://carla.readthedocs.io/en/latest/tuto_D_create_semantic_tags/

closed 2024-11-06 by Blyron