carla-simulator/carla #4941: Is there any way to install and run CARLA in a vmware (ubuntu) on Windows ?

aslirey, 2021-12-02:
I have tried to install Carla in vmware workstation 16 on Windows. Ubuntu's version is 18.04. 
[image]

MattRowe18, 2021-12-02:
@aslirey running CARLA within a VM can be tricky, since CARLA requires high performance graphics and desktop VM solutions often abstract the graphics hardware with intermediary driver layers. I would suggest to look for ways to allow the VM to have direct access to the host GPU, bypassing intermediary steps introduced by the virtualisation solution. 

Research on Google about "PCI passthrough" for VMware. 

Also as an alternative, CARLA works well in Windows, you could consider installing and running CARLA on your Windows host. If there is some reason you need to use the Python API within your Ubuntu VM, you could still achieve this since CARLA is designed for clients to access the CARLA server through a network, you can access your host machine as if it were a network machine, it would need configuration, but it can be done.

closed 2021-12-02 by MattRowe18

aslirey, 2021-12-02:
@MattRowe18 Thanks for your fast answer.

TauJiang, 2022-10-26:
But I have an additional question which is that I want to use ROS bridge, but it is not available on Windows, so I have to try VM, but since VM is not working I guess I have just to make a second system

SruthiSreeSG, 2024-11-13:
Hi! I try to install Virtual Box in Mac M3. But I didn't get windows option in type section to install windows VM. I have downloaded Windows .iso ile. But i don't know how to set the Virtual Machine. Can anyone help me out?

Also please tell me is it possible to download and use Carla simulator in Mac? If yes please provide me the steps to install them