carla-simulator/carla #4942: get  vehicle's ground truth

whuxhzhang, 2021-12-02:
I found that there are some vehicles on the Town04 map. How can I get the ground truth of these vehicles?

[image]

MattRowe18, 2021-12-02:
@whuxhzhang hi

Can you expand on what you need as the "ground truth"?

whuxhzhang, 2021-12-03:
> @whuxhzhang hi
> 
> Can you expand on what you need as the "ground truth"?

thank you for your reply，I have found a solution.
bboxes = world.get_level_bbs(carla.CityObjectLabel.Vehicles)
[https://github.com/carla-simulator/carla/issues/2343](url)

whuxhzhang, 2021-12-03:
https://github.com/carla-simulator/carla/issues/2343
There is already a solution here.

closed 2021-12-03 by whuxhzhang