carla-simulator/carla #4952: How to get the velocity of the ego vehicle?

vikram0230, 2021-12-07:
I have checked the carla docs but couldn't figure how to get the velocity of the vehicle. Later, I found that there is a workaround, by calculating the distance travelled from the location and dividing it by the time travelled. 

Is there a better way to get the velocity of the ego?

JochemTSR, 2021-12-08:
The velocity of any actor can be retrieved through the [`get_velocity` method.](https://carla.readthedocs.io/en/0.9.13/python_api/#getters) This gives you a 3D vector containing the velocity in each dimension. If you want the total velocity you can take the square root of the sum of the squared components: sqrt(v<sub>x</sub><sup>2</sup>+v<sub>y</sub><sup>2</sup>+v<sub>z</sub><sup>2</sup>)

closed 2021-12-08 by vikram0230

VidaAPI, 2023-06-19:
> The velocity of any actor can be retrieved through the [`get_velocity` method.](https://carla.readthedocs.io/en/0.9.13/python_api/#getters) This gives you a 3D vector containing the velocity in each dimension. If you want the total velocity you can take the square root of the sum of the squared components: sqrt(vx2+vy2+vz2)

Hi @JochemTSR Can i ask if you managed to retrieved the velocity of the adversary (in follow leading vehicle) and plot them with respect to time. Since  couldn't find such thing in metrics, I'm trying to log into file. but nothing is being recorded