carla-simulator/carla #4953: tutorial.py error

aslirey, 2021-12-07:
Hello everyone,

it gave  the error below when I executed tutorial.py example. 

PS C:\Program Files\carla\PythonAPI\examples> python tutorial.py
created vehicle.ford.crown
INFO:  Found the required file in cache!  Carla/Maps/TM/Town10HD_Opt.bin
created sensor.camera.depth
moved vehicle to Location(x=84.055626, y=-60.723831, z=0.398577)
created vehicle.micro.microlino
Traceback (most recent call last):
  File "tutorial.py", line 90, in <lambda>
    camera.listen(lambda image: image.save_to_disk('_out/%06d.png' % image.frame, cc))

How can I fix it? Thanks in advance.

aslirey, 2021-12-07:
When I ran a Command Prompt as administrator, The error went away.

closed 2021-12-07 by aslirey

9527huang, 2022-04-01:
hello , do you fix it?