carla-simulator/carla #4963: CarSim manual control example not working

svanimisetti, 2021-12-11:
I am unable to use CarSim plugin example in CARLA as per [instructions](https://carla.readthedocs.io/en/latest/tuto_G_carsim_integration/#before-you-begin) in the documentation. The following information should help reproduce the issue.

CARLA version: 0.9.13
Platform/OS: Windows 10
Problem you have experienced: *manual_control_carsim.py does not use CarSim vehicle model physics*
What you expected to happen: CARLA to use CarSim model for physics - but it does not.

### Steps to reproduce:

1. Download [latest version](https://carla-releases.s3.eu-west-3.amazonaws.com/Windows/CARLA_0.9.13.zip) of CARLA and install to C:\CARLA
2. CarSim program installed to C:\CarSim2021p1_prog. Open program and export UE4 example (Run Control dataset -> External Control, Wrapper -> Unreal Engine 4 Plug-in, Lead Vehicle) to C:\CarSim2021p1_data\ue4simfile.sim
3. Start CarSim GUI or license server using `C:\CarSim2021p1\Programs\cs-lm-cli.exe` to acquire solver license.
4. Download [CarSim UE4 plugin v2021.1 r166435](https://www.carsim.com/users/download2.php?f=CarSim_UE4.27_Plugin_v2021.1.2_2021-09-22_r166435-web.zip) from MSC website and unzip the contents to C:\CARLA\CarlaUE4\Plugins\CarSim
5. Start CARLA server using `cd C:\CARLA && CarlaUE4.exe -windowed -carla-server -carla-no-networking -quality-level=Low`
6. Modify line 419 in `manual_control_carsim.py` to:
> `world.player.enable_carsim("C:/CarSim2021p1_data/ue4simfile.sim")`
7. Start CarSim example provided with CARLA installation using `cd C:\apps\CARLA\PythonAPI\examples && 
python manual_control_carsim.py`
8. The pygame console starts - and I can control the vehicle. However, I think the solver is not being used, as the solver output and log files are not being written to the CarSim results folder for this testrun (`C:\CarSim2021p1_data\Results\Run_dd7a828d-4b14-4c77-9d09-1974401d6b25`)

Do I need to compile CARLA from source on my computer to be able to run the CarSim example? This is not clearly mentioned in the documentation. If re-compilation is not needed, it would be helpful if the steps to install the CarSim plugin into CARLA installation are clearly explained.

woojmn, 2022-01-07:
Hello,
I have same problem, too.

When I print the output of world.player.enable_carsim("C:/CarSim2021p1_data/ue4simfile.sim") after modifying the line 419 in 'manual_control_carsim.py', the output is Nonetype.

Can you solve that problem??

crabiner, 2022-03-13:
Hello,
I had a similar experiance of crash in carsim plugin using carla 0.9.13 compiled from source and [VehicleSim_Plugin-Example-4.26_Project-r154045.zip](https://www.carsim.com/users/download2.php?f=VehicleSim_Plugin-Example-4.26_Project-r154045.zip) with the sim file provided at [VehicleSim_Plugin-Example-4.26_Project-r154045.zip](https://www.carsim.com/users/download2.php?f=VehicleSim_Plugin-Example-4.26_Project-r154045.zip)
I could see carla output saying loading library and immediately saying releasing library :

Loading simfile: .../ue4simfile.sim
Loading library: /tmp/VSSuxK9E7
Releasing library

Did anyone have a working experiance of carla + carsim on ubuntu 18.04?

shengkaicui, 2022-04-15:
> 

hi, i'm using Linux version of both and Carsim not working either. On your procedure 8, you mentioned pressed k, did you try to press CTRL + k? Because in the script it says "**event.key == K_k and (pygame.key.get_mods() & KMOD_CTRL)**"

jy723, 2022-04-19:
The problem has been solved??I have the same problem.

YSFlower, 2022-08-16:
I followed instructions to install the plugin,but when I run `manual_control_carsim.py`, pressing 'ctrl+k' to load '.sim' file; An error came: Unable to create Vehiclesim Vehicle. Error message from VS Solver:Failed to copy solver DLL file "" to temporary location.
Has anyone met same problem?

sushil-bhatta, 2022-12-19:
Hi @crabiner, are you able to solve this issue ? I am also having this problem and could not figure out, how to deal with it.
If you have any progress on it, could you please share some information.

szyvcdc, 2023-04-23:
same problem, and unable to solve this issue, by far.

Pt0lemaeus, 2023-11-02:
could someone send me the plugin file and sim file？t.xm@foxmail.com 
I want to have a try