carla-simulator/carla #4964: Town06 and Town07 'map not found'

apiszcz, 2021-12-11:
CARLA version: 0.9.13/
Platform/OS: Windows 10
Problem you have experienced: Unable to load Town06 and Town07, all others load
What you expected to happen: 'map not found'
Steps to reproduce: python config.py --map Town06

apiszcz, 2021-12-11:
Issue resolved
python config.py --list
    Town01, Town01_Opt, Town02, Town02_Opt, Town03, Town03_Opt,
    Town04, Town04_Opt, Town05, Town05_Opt, Town10HD, Town10HD_Opt.

closed 2021-12-11 by apiszcz

varunjammula, 2022-01-27:
Same issue.

aerofentse, 2022-08-05:
Map not found issue still present

VipulRamtekkar16, 2022-09-05:
To add the map of Town06 and Town07 download and extract the Additional maps zip file (https://carla-releases.s3.eu-west-3.amazonaws.com/Windows/AdditionalMaps_0.9.13.zip (windows file)) and extract the file and run ImportAssets.sh

VipulRamtekkar16, 2022-10-21:
When you extracted a precompiled version of carla it was already present. I am attaching my file for reference its for ubuntu check if the versions match you could use it in that case.

VipulRamtekkar16, 2022-10-21:
I extracted from the release files of carla so it was present in that. I am using ubuntu system so I am not sure but I expect windows to have something similiar. I found the file on extracting from this zip: https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/CARLA_0.9.11.tar.gz

shyam1494, 2022-10-29:
> Where is ImportAssets.sh? I do not see it in the Import folder.

Hi Apurva , 
I am also facing the same issue, Could you able to solve it

ajdroid, 2022-11-01:
It's in the Utils folder