carla-simulator/carla #4965: spawn_npc missing?

apiszcz, 2021-12-11:
CARLA version: 0.9.13
Platform/OS: Windows 10
Problem you have experienced:
What you expected to happen: spawn_npc.py missing from examples directory

apiszcz, 2021-12-11:
should the documentation reflect the new generate_traffic.py?

aslirey, 2021-12-11:
I too use Carla on Windows. spawn_npc.py is within carla\Co-Simulation\Sumo directory for me. I think you may look at the same directory. However, I couldn't execute it and encountered with an error. If you make it execute, could you please inform me?
Error: C:\Users\re\anaconda3\python.exe: can't open file 'spawn_npc.py': [Errno 2] No such file or directory

svanimisetti, 2021-12-11:
@apiszcz - I stumbled across exactly the same issue last week when I upgraded to 0.9.13. generate_traffic.py seems to have the same functionality as spawn_npc.py - so I am assuming that it is a replacement. And yes, the documentation does not seem to reflect this yet.

jackbart94, 2021-12-14:
Probably you've been checking an older version of the documentation. As far as I know, almost all instances of ``spawn_npc.py`` mentioned there have been replaced with the new ``generate_traffic.py`` script. It's basically the same, just a name change. We'll double check the documentation anyway, thanks for mentioning this!

closed 2021-12-14 by jackbart94

ricardodeazambuja, 2022-02-15:
Just to flag it here, the documentation (0.9.13 and latest) is still using the `spawn_npc.py`: 
https://carla.readthedocs.io/en/latest/tuto_G_retrieve_data/#carla-traffic-and-pedestrians

jackbart94, 2022-02-15:
@MattRowe18 Please, can you make this change?