carla-simulator/carla #4966: CARLA crashes when reloading world with low quality settings

svanimisetti, 2021-12-11:
CARLA version: 0.9.13
Platform/OS: Windows 10
Problem you have experienced: CARLA crashes when reloading world with low quality settings

#### Steps to reproduce:

1. Start windows command-prompt and launch CARLA 0.9.13 (installed to C:\CARLA) using following two steps (had to wait ~30secs between two commands for my computer configuration).

> `start "" /D C:\apps\CARLA /MIN /B CarlaUE4.exe -windowed -carla-server -carla-no-networking -quality-level=Low`
> `start "" /D C:\apps\CARLA\PythonAPI\util /MIN /B python config.py --map Town01`

2. Start Python and run following

> `import os, sys`
> `sys.path.append(r'C:\apps\CARLA\PythonAPI\carla\dist\carla-0.9.13-py3.7-win-amd64.egg')`
> `import carla`
> `client = carla.Client('localhost', 2000)`
> `client.reload_world()`

3. Above causes CARLA to crash with following error written to python terminal
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> RuntimeError: time-out of 5000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
[image]

4. Remove `-quality-level=Low` in Step 1 and launch CARLA again
   
5. Now run commands in Step 2, and CARLA should work fine (warning message in Step 3 still an issue).

crisdesivo, 2022-12-17:
I have the same issue on ubuntu 20 and carla 0.9.13

IAMHAADICOOL, 2025-04-22:
This is a serious issue. Please, developers, consider resolving this it. It would be beneficial to an entire community.