carla-simulator/carla #4969: CARLA netconvert_carla.py not converting .xodr to .net.xml for SUMO >1.8.0

svanimisetti, 2021-12-13:
CARLA version: 0.9.13
Platform/OS: Windows 10
Problem you have experienced: *netconvert_carla.py is fails to convert OpenDRIVE files*
What you expected to happen: *SUMO .net.xml files should have been created - but there is an error instead.*

### Steps to reproduce:

1. CARLA for Windows setup in `C:\apps\CARLA`
2. SUMO setup in `C:\apps\sumo-1.9.0`
3. Create following test script at `%USERPROFILE%\Desktop\convert_xodr.py

```Python
import os, sys, glob

os.environ['CARLA_HOME'] = 'C:/apps/CARLA'
os.environ['SUMO_HOME'] = 'C:/apps/sumo-1.9.0'

egg_file = glob.glob(os.path.join(os.getenv('CARLA_HOME'),
    'PythonAPI/carla/dist/carla-*%d.%d-%s.egg' %
    (sys.version_info.major, sys.version_info.minor,
    'win-amd64' if os.name == 'nt' else 'linux-x86_64')))[0]
sys.path.append(egg_file)
import carla

sys.path.append(os.path.join(os.getenv('SUMO_HOME'),'tools'))
os.environ['PATH'] = os.pathsep.join(
    [os.path.join(os.getenv('SUMO_HOME'),'bin'),os.getenv('PATH')])
    
carla_sumo_dir = os.path.join(os.getenv('CARLA_HOME'), 'Co-Simulation/Sumo')
if os.path.isdir(carla_sumo_dir):
    sys.path.append(carla_sumo_dir)

from util.netconvert_carla import netconvert_carla

xodr_file = os.path.join(os.environ['USERPROFILE'],'Desktop','Town01.xodr')
net_file = os.path.join(os.environ['USERPROFILE'],'Desktop','Town01.net.xml')
netconvert_carla(xodr_file, net_file, guess_tls=True)
```

4. Copy `Town01.xodr` to Desktop from `C:\apps\CARLA\CarlaUE4\Content\Carla\Maps\OpenDrive\Town01.xodr`
5. Run `cd %USERPROFILE%\Desktop && python convert_xodr.py` and it produces following error.

```
proj_create: Error -38 (failed to load datum shift file): vgridshift: could not find required grid(s).
Error: Could not set projection. (Could not build projection!)
Quitting (on error).
Traceback (most recent call last):
  File "convert_xodr.py", line 25, in <module>
    netconvert_carla(xodr_file, net_file, guess_tls=False)
  File "C:/apps/CARLA\Co-Simulation/Sumo\util\netconvert_carla.py", line 521, in netconvert_carla
    _netconvert_carla_impl(xodr_file, output, tmpdir, guess_tls)
  File "C:/apps/CARLA\Co-Simulation/Sumo\util\netconvert_carla.py", line 394, in _netconvert_carla_impl
    raise RuntimeError('There was an error when executing netconvert.')
RuntimeError: There was an error when executing netconvert.
```

6. Now change `line 4` in  `convert_xodr.py` to `os.environ['SUMO_HOME'] = 'C:/apps/sumo-1.8.0'` or any SUMO version <=1.8.0.
7. Re-run `convert_xodr.py` and this time it converts the .xodr without any issue.

This seems to be an issue with newer version of SUMO - which may be using some new specifications for OpenDRIVE. Not sure, but the issue is reproduced in multiple machines.

ik0015, 2022-01-05:
I cant understand how to resolve this issue. I'm attaching the error screenshot for your further help.
best,
[image]

namdre, 2022-10-10:
For the record: Updating sumo should help.

SExpert12, 2023-06-04:
CARLA : 0.9.14
Sumo - 1.17.0
OS-Ubuntu 20.04 ( focal)
~/Downloads/CARLA_0.9.14/Co-Simulation/Sumo/util$ python3 netconvert_carla.py Town05.xodr 
Error: File 'Town05.xodr' is not accessible (No such file or directory).
Error: No nodes loaded.
Quitting (on error).
Traceback (most recent call last):
  File "netconvert_carla.py", line 541, in <module>
    netconvert_carla(args.xodr_file, args.output, args.guess_tls)
  File "netconvert_carla.py", line 521, in netconvert_carla
    _netconvert_carla_impl(xodr_file, output, tmpdir, guess_tls)
  File "netconvert_carla.py", line 394, in _netconvert_carla_impl
    raise RuntimeError('There was an error when executing netconvert.')
RuntimeError: There was an error when executing netconvert.

I have the same problem with latest version of CARLA and Sumo simulator both.

How to resolve this?

namdre, 2023-06-04:
Please try calling netconvert directly according to the options found in `netconvert_carla.py`:

`netconvert --opendrive-files yourfile.xodr --output-file sumo.net.xml --geometry.min-radius.fix --geometry.remove --opendrive.curve-resolution 1 --opendrive.import-all-lanes --output.original-names  --tls.discard-loaded true`

also, check whether you are using 1.17.0 by calling `netconvert` without arguments which should report the version number.

SExpert12, 2023-06-05:
Thank you sir for your reply. I would try this.