carla-simulator/carla #4974: Inconsistent Radar and carla.World.project_point() detections

JoAMichael, 2021-12-13:
**CARLA version:** 0.9.12
**Platform/OS:** Ubuntu 20.04.3 LTS

**Problem you have experienced:**

When using the standard radar sensor from Carla, inconsistencies in detections may be observed.

### 1.  Some Buildings are not detected
[image] 
[image] 

This for Town03, spawn point index 1.

The red and black detections result from the radar itself as well as `world.project_point()`, respectively.
Two of the buildings seem to be invisible to the radar as well as `project_point()`, with detections appearing right behind it from the sensors point of view.

### 2. Detections may clip moving objects resulting in wrong speed
Consider the simple scenario depicted below, with a car driving straight away from the ego vehicle with radar.
[image]
Here's an excerpt from the radar output CSV file:
[image]
Even though both points seem to be at almost the same location and the object hit by the radar trace (`hit_obj`) is a vehicle in both cases the velocities are radically different. Once we have the correct value of 5m/s and once 0m/s.
The hit_obj entries were obtained using the following code:
```python
# This maps the local detect_direction to the global coordinates and direction for the detection
detect_direction = carla.Vector3D(x=1)
carla.Transform(carla.Location(),
                          carla.Rotation(pitch=radar_rot.pitch + alt, yaw=radar_rot.yaw + azi,
                                                 roll=radar_rot.roll)).transform(detect_direction)
# Recasting a ray from sensor to detected location, but getting object this time
hit_point = self.carla_world.project_point(current_loc, detect_direction, depth + 1)
return hit_point
```
With `hit_point` being a `carla.LabelledPoint` and with the hit object as `hit_point.label`.
So, the detection is clearly in line with the vehicle, otherwise `project_point()` would not return a `LabelledPoint` with label 'Vehicles'. 
However, the altitude angle is lower for the detection with the wrong speed, making me think that the ground was actually detected by the radar here (the only object in the scene with velocity = 0).

Here is another excerpt from the same CSV:
[image]

Again, `project_point()` tells us that the first object along the line of sight is a vehicle, in both cases.
However, this time not only the velocities but also the ranges are radically different. 
This shows clearly that either the radar traces sometimes pass through moving objects or that `project_point()` does not return the proper `LabelledPoint`.

I am not sure if this is the same issue as with the invisible buildings, as not a single detection hits the buildings above but most times the car is detected.

**What you expected to happen:** That the radar properly detects objects and their speed in Carla.
**Steps to reproduce:** 

1. Place car at spawn point 1 in Town03 with radar in front. Use `carla.DebugHelper.draw_point()` to visualize detections.
2. Do a simple recording of a moving actor with the radar and observe issues in the resulting CSV file.

**Other information (documentation you consulted, workarounds you tried):**
I am by no means an expert, but perhaps the issues with dynamic objects are due to interpolation when the trace hits the sub-pixel range on the corner of an object. 
This is explained nicely in a paper [here](https://arxiv.org/abs/1905.00160), specifically section 3.B.

sbv198, 2022-06-27:
@JoAMichael 
I am using CARLA 0.9.13 and having similar issue with radar  Sometimes it detects object sometimes it does not in some cases i had velocity around 900 while trying a scenario with pedestrian coming across vehicle

sbv198, 2022-06-27:
@JoAMichael 
discrepencies are still there in the results from radar points but what you can do is 
[image]
when the vehicle is moving from  away from targt vehicle you can try to take only positive velocities.

[image]

when both vehicles are moving in opposite direction you will see that all the velocities are in negative so what you can do is generate some algorithm to avoid the points which have same velocities of ego vehivle

 list_test = []
            for i in range(len(points)):
                #if abs(round(points[i][0])+round(vel)) <= 1.0:
                if (points[i][0])<= 0.0:
                    pass
                else:
                    list_test.append(points[i].tolist())
            print("list_test", list_test)
i am uysing this algo rithm to avoid the points that i do not need 

hope that might help you

JoAMichael, 2022-07-07:
Hey @sbv198 ! 

Yes you are right that filtering is more of a band aid for specific scenarios than an actual solution.
In my case I would like to use the radar for all kinds of stuff, so I wouldn't always know the exact setting/positioning/velocities of all vehicles. Of course Carla can provide all this information, but computing all that and filtering out the erroneous radar data is almost as much effort as just making a new radar sensor yourself.

There are some glaring issues with the radar sensor and many missing "features" such as giving information about received power depending on materials and/or distance, proper ray tracing (with reflections) or micro-doppler, which are absolutely essential for radar sensors in autonomous driving. 
@DSantosO had previously assured me (in the Carla Discord) that changes are on the roadmap, but it'll probably be a while before they're implemented.

You have to understand that (IMO) Carla seems to be driven by mainly ML/AI focused approaches for autonomous driving.
These approaches rely mainly on visual data and as you can see that is where Carla offers the most (many different types of camera sensors). 
Of course you can also feed radar data to a neural net but that doesn't seem to be the hype at the moment (see Tesla).

I would be careful using the Carla radar sensor in its current state as it will probably change significantly in the future.

Also I do not see solutions to the problems you or I mentioned, as they are bugs and probably require changes in the source code.

sbv198, 2022-07-08:
hey @JoAMichael 

what you said is true.

In my case my whole work has been done using radar sensor and i do not have enough time to learn much in details about how to implement different camera sensors for data retrieving

xu24889070, 2023-06-01:
hello,i have researhed these issuses long time.To be honest,Carla is not good at radar, whcih includes provide incorrect  and randome velocity,moreover,the velocity only is relative of ego vehicle.RSC is not provided in Carla, i think this does confused my resarch.if you have any question , you can email me to xcc6219@163.com,so that i can answer you in time.

sumandasntu, 2023-08-24:
Hi, please help me with the definition of current_loc.

yourj4m, 2023-08-24:
> Hi, please help me with the definition of current_loc used in carla_world.project_point().

Hey @sumandasntu, the `current_loc` is just the location of the sensor doing the recording.  
In this case, if you have the sensor data as a `carla.SensorData` object, you can get the location of the sensor by using `carla.SensorData.transform.location`.
You have to pay close attention to the coordinates though, since if the sensor is attached to a vehicle you might get local coordinates.

laduflot, 2025-08-11:
> @JoAMichael
> 
> yes my bad i forgot to annotate the points, i think you are right about what you have said i also don' t get this kind of velocities after the simulation has started running
> 
> def rad_callback(radar_data):
> 
> ```
>         current_rot = radar_data.transform.rotation
>         for detect in radar_data:
>             azi = math.degrees(detect.azimuth)
>             alt = math.degrees(detect.altitude)
>             # The 0.25 adjusts a bit the distance so the dots can
>             # be properly seen
>             fw_vec = carla.Vector3D(x=detect.depth - 0.25)
>             carla.Transform(
>                 carla.Location(),
>                 carla.Rotation(
>                     pitch=current_rot.pitch + alt,
>                     yaw=current_rot.yaw + azi,
>                     roll=current_rot.roll)).transform(fw_vec)
>         points = np.frombuffer(radar_data.raw_data, dtype=np.dtype('f4'))
>         points = np.reshape(points, (len(radar_data), 4))
>         #print('points :', points)
> ```
> 
> i have used this code provided in carla documents from that document i came to know that this array represent velocity azimuth, altitude and distance respectively

I suspect that this code, I mean the position of fw-vec, is not working when the current_rot.roll is not 0 (the most obvious is to test with 180 degree, It feel like the drawn point don't have the right rotation. 
If you force alt, azi value and test this code with current_rot.roll = 0 or 180, you will that the value of fw is not changing. 

I don't understand that. Is this piece of code relevant ?