carla-simulator/carla #4975: Walkers are not part of CARLA replay

ptr-br, 2021-12-13:
Hello,

I'm quite new to CARLA and encounter some issues with walkers when replaying files.
More specifically, it seems like walkers and their movements are not stored completely (if at all) when using the API client recorder.

I first encountered the issue when replaying some files with my own code. The walkers movement seemed less smooth and their relative body positions did not completely match their translation direction.

I first thought, that this could be due to my implementation and tried to see if the behavior is also reproducible using CARLA examples.
And indeed, when using the `manual_control.py` script, pedestrians do not behave as I thought they would. They somehow seem to flicker when the file is replayed. Furthermore, they are not reset  when the scene playback is triggered, but simply move on.

Is this an intended behavior?
If yes, is there any workaround to also replay the walkers in an accurate fashion?

Steps to reproduce (using example scripts):
- generate traffic using trafffic-generator :  `python3 generate_traffic.py -w 100`
- start manual control script: `python3 manual_control.py`
- and record some scene with walkers  using `ctrl+R`
- replay scene using `ctrl+P`

CARLA version: 0.9.13 - (Quick-start and Build from Source) | also tried CARLA version 0.9.12 
Platform/OS: Ubuntu

Other information (documentation you consulted, workarounds you tried): 
- Set additional_data=True for client.start_recorder()

uservinu, 2021-12-14:
@ptr-br  I have also faced the same problem. Thought it was my script. My workaround is to do the replay in a separate session.(start a new client, start replay, wait till the recorded actors are spawned, attach cameras to them and start saving data, in sync mode)

ptr-br, 2021-12-14:
Thank you for the response @uservinu.

Does everything look good with your workaround? 
Because I also tried to do that, and as described the pedestrians then seem to have some artifacts and their movement is not as smooth (looks like trajectory and body movements do not seem to fit perfectly).
Actually I found out that the replayed file looks good when using the replay example script `start_replaying.py` and watching it with the spectator...
So there might be some issue in my implementation as well...

However, the behavior of the example scripts for replaying the scene seems weird to me...

uservinu, 2021-12-14:
[image]
[image]
Upon close inspection of the recorded images I also have some artefacts on the pedestrians, they seems to be more pronounced when they are further away from the camera and are running. You can see that in the attached images(check the yellow jacket guy)

ptr-br, 2021-12-16:
Ok the simplest example for this problem I could come up with is as follows:

-  comment out line 493 in manual_control.py since walker does not have an auto-pilot...
` # world.player.set_autopilot(self._autopilot_enabled)`
- run the script as a walker:
`python manual_control.py --filter walker.*`
- record and replay scene

The same problems as mentioned above can be seen (walker has artifacts & walking direction and body position do not match).

I hope this is not due to my local dependencies so it would be great if some could reproduce the issue...

zsxzs, 2022-11-25:
> [image] [image] Upon close inspection of the recorded images I also have some artefacts on the pedestrians, they seems to be more pronounced when they are further away from the camera and are running. You can see that in the attached images(check the yellow jacket guy)

@uservinu  Sorry to bother you. your workaround looks good. but I have no idea how to use camera attached vehicle to collect rgb images when replay the recorder file.  Could you give me a hint? Thank you very much.

closed 2024-10-17 by Blyron