carla-simulator/carla #4980: Walker bone transforms are not in sync with the rgb images

uservinu, 2021-12-14:
CARLA version: 0.9.13
Platform/OS: Windows
Problem you have experienced: I used the code from the examples/draw_skeleton.py to draw the bone positions of pedestrians. The bone positions seems to be out of sync with the rgb images  for pedestrians who are running. 
[image]

What you expected to happen: Bones in sync with the images.
Steps to reproduce: I am using sync mode with fixed delta secs  and the simulation is replaying a recording.

uservinu, 2021-12-15:
```
#!/usr/bin/env python
# Copyright (c) 2021 Computer Vision Center (CVC) at the Universitat Autonoma de
# Barcelona (UAB).
#
# This work is licensed under the terms of the MIT license.
# For a copy, see <https://opensource.org/licenses/MIT>.

import glob
import os
import sys
import math
import argparse
import copy
import time
from multiprocessing import Pool
from PIL import Image

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla
import random

try:
    import pygame
except ImportError:
    raise RuntimeError('cannot import pygame, make sure pygame package is installed')

try:
    import numpy as np
except ImportError:
    raise RuntimeError('cannot import numpy, make sure numpy package is installed')

try:
    import queue
except ImportError:
    import Queue as queue

class CarlaSyncMode(object):
    """
    Context manager to synchronize output from different sensors. Synchronous
    mode is enabled as long as we are inside this context

        with CarlaSyncMode(world, sensors) as sync_mode:
            while True:
                data = sync_mode.tick(timeout=1.0)

    """

    def __init__(self, world, *sensors, **kwargs):
        self.world = world
        self.sensors = sensors
        self.frame = None
        self.delta_seconds = 1.0 / kwargs.get('fps', 20)
        self._queues = []
        self._settings = None

    def __enter__(self):
        self._settings = self.world.get_settings()
        self.frame = self.world.apply_settings(carla.WorldSettings(
            no_rendering_mode=False,
            synchronous_mode=True,
            fixed_delta_seconds=self.delta_seconds))

[... 340 lines cut ...]

                # display.blit(font.render('%d bones' % len(points), True, (255, 255, 255)), (8, 10))

                pygame.display.flip()
                time.sleep(0.1)

    finally:
        # time.sleep(5)
        print('destroying actors.')
        for actor in actor_list:
            actor.destroy()
        pygame.quit()
        pool.close()
        print('done.')

if __name__ == '__main__':

    try:

        main()

    except KeyboardInterrupt:
        print('\nCancelled by user. Bye!')

```

ptr-br, 2021-12-16:
Hi, i also encounter some issues when drawing the skeletons into recordings...

I replayed a file and took a screenshot of the **spectator** at a certain frame which looks like this:
[image]
_Note: Screenshot is taken with a screen saving tool since i do not now a CARLA method to get the spectator output..._

A **rgb** camera I placed in the scene (at the same position as the spectator) gives me this result:
[image]

Drawing the **bones** into the **rgb** image (same frame) looks like this:
[image]

Let's take a closer look at the left walker in the three pictures.
Here again the three scenes but only containing the part I want to focus on:

**spectator**:
[image]

**rgb**:
[image]

**rgb with bones**:
[image]

When looking at the images closely there are a few things to notice:

- The walkers contains some artifacts in the rgb image compared to the spectator screenshot 
- The walker on the rgb image seems to have a slight rotation compared to the one captured with the spectator. Let's look closely on the left foot. One can see, that on the spectator image the foot is slightly touching the next crosswalk stripe. On the rgb image this is not the case (because of some rotation I guess).
- These issues are already partly mentioned in https://github.com/carla-simulator/carla/issues/4975
- Interestingly, this issue also persists when drawing the skeletons into the rgb image, causing them them to not align properly. If we look at the skeleton on the third image we can see that it is indeed above the next stripe of the crosswalk (as the spectator screenshot suggests), but does (of course) not match up with the walkers feet on the rgb image...

Therefore I think that the bone positions seem to be correct, but the walker displayed in the rgb image is not...

@uservinu are you only encountering this issue for running pedestrians, or for all walkers in replayed files as i do?

uservinu, 2021-12-16:
@ptr-br  I have the issue for all pedestrians, just the issue is very bad for running pedestrians.

PR #5987 (2022-11-29): Recorder with skeleton bones
#### Description

The recorder now can store the bone position of the pedestrian skeleton. We need to enable the additional data flag for the recorder. Then the replayer will pose the pedestrians at the exact pose each frame.

Fixes #4980, #4975

#### Where has this been tested?

  * **Platform(s):** windows
  * **Python version(s):** 3.7
  * **Unreal Engine version(s):** 4.26

#### Possible Drawbacks

Recorder files a bit bigger due to the increment of data

- - -
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/5987)
closed 2023-06-02 by stale[bot]