carla-simulator/carla #4982: Custom Sensor - fails to compile when running "make rebuild"

Piyush54321, 2021-12-14:
Hello, I am having issues to compile with custom sensor.

CARLA version: 9.13
Platform/OS: Windows 10

Fails to compile when adding custom sensor custom sensor. I am getting this error "source/libcarla/libcarla.cpp(7): fatal error C1083: Cannot open include file: 'carla/Memory.h': No such file or directory". If I try to fix this by providing the path to 'Memory.h' the error progresses to find other missing/cannot open files within 'Memory.h' and keep throwing out similar errors

Log file where error appear: No errors up to this point

"""osm2odr.vcxproj -> C:\Carla\carla\Build\osm2odr-visualstudio\src\Release\osm2odr.lib
  -- Install configuration: "Release"
  -- Installing: C:/Carla/carla/PythonAPI/carla/dependencies/lib/osm2odr.lib
  -- Installing: C:/Carla/carla/PythonAPI/carla/dependencies/include/OSM2ODR.h
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\Carla\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --rebuild
-[BuildPythonAPI]: Cleaning "C:\Carla\carla\PythonAPI\carla\build"
-[BuildPythonAPI]: Cleaning "C:\Carla\carla\PythonAPI\carla\source\carla.egg-info"
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
creating source\carla.egg-info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
writing manifest file 'source\carla.egg-info\SOURCES.txt'
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
creating build
creating build\lib.win-amd64-3.7
creating build\lib.win-amd64-3.7\carla
copying source\carla\command.py -> build\lib.win-amd64-3.7\carla
copying source\carla\__init__.py -> build\lib.win-amd64-3.7\carla
running build_ext
building 'carla.libcarla' extension
creating build\temp.win-amd64-3.7
creating build\temp.win-amd64-3.7\Release
creating build\temp.win-amd64-3.7\Release\source
creating build\temp.win-amd64-3.7\Release\source\libcarla
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\PPAIRAIK\AppData\Local\Programs\Python\Python37\include -IC:\Users\PPAIRAIK\AppData\Local\Programs\Python\Python37\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.7\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
source/libcarla/libcarla.cpp(7): fatal error C1083: Cannot open include file: 'carla/Memory.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\Carla\carla\PythonAPI\carla\dist"!

C:\Carla\carla>""""

Thanks for your help.
-->

myleskeller, 2023-02-19:
I'm experiencing the exact same issue with Carla 0.9.14 on Windows 10.
As far as I can tell, `make rebuild` is completely broken on 0.9.14 (and was likely as well on 0.9.13) for Windows, because it doesn't even function correctly for me on a **fresh** install. 
I saw somewhere that if you run `make clean`, `make PythonAPI`, and `make launch`, you are effectively doing the same thing. So I did that, but got the same `error C1083` at `make PythonAPI`. I ran `make PythonAPI` again, without changing literally anything, and it _succeeded_ this time? Following that, `make launch` worked as well.
**That was all on the fresh install.**

So I tried applying that to our current situation (I just finished implementing the SafeDistanceSensor tutorial) and substituted `make rebuild` with the process outlined above. Still got to the same error `error C1083` after the first `make PythonAPI`, but the second attempt yields a slightly different error:

```
d:\carla\carla>make PythonAPI
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  32
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "d:\carla\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "d:\carla\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "d:\carla\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "d:\carla\carla\Build\" --zlib-install-dir "d:\carla\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "d:\carla\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "d:\carla\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "d:\carla\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "d:\carla\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "d:\carla\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "d:\carla\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "d:\carla\carla\Build\recast-0b13b0-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "d:\carla\carla\Build\" --toolset msvc-14.2 --version 1.80.0 -j 32
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "d:\carla\carla\Build\boost-1.80.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "d:\carla\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "d:\carla\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
        1 file(s) copied.
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "d:\carla\carla\Build\"
    -[install_sqlite3]: A Sqlite3 installation already exists.
    -[install_sqlite3]: Delete "d:\carla\carla\Build\sqlite3-install" if you want to force a rebuild.
    -[install_sqlite3]: Exiting...
        1 file(s) copied.
-[Setup]: Installing PROJ
    -[install_proj]: [Batch params]: --build-dir "d:\carla\carla\Build\"
    -[install_proj]: A PROJ installation already exists.
    -[install_proj]: Delete "d:\carla\carla\Build\proj-install" if you want to force a rebuild.
    -[install_proj]: Exiting...
        1 file(s) copied.
-[Setup]: Installing Eigen
    -[install_eigen]: [Batch params]: --build-dir "d:\carla\carla\Build\"
0 File(s) copied
0 File(s) copied
        1 file(s) copied.
        1 file(s) copied.

    -[install_eigen]: Eigen has been successfully installed in "d:\carla\carla\Build\eigen-install"!
    -[install_eigen]: Exiting...
[... 87 lines cut ...]
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\Myles\AppData\Local\Programs\Python\Python38\include -IC:\Users\Myles\AppData\Local\Programs\Python\Python38\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\M [... cut ...]
source/libcarla/libcarla.cpp(7): fatal error C1083: Cannot open include file: 'carla/Memory.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "d:\carla\carla\PythonAPI\carla\dist"!
```

Not sure if this is progress or not, but it seems like the error persists in 0.9.14, even when using other attempted workarounds.

FlexiQQ, 2023-02-24:
Actual! 
Same issue!

Lukas-Dierkes, 2023-03-23:
Same issue for me. Is there any fix ongoing?

ronyshaji, 2023-06-15:
@myleskeller Did you find a solution to the issue carla/memory.h not found?

alimuratov, 2024-10-02:
@ronyshaji Hey, have you found a solution?

chanyca, 2024-11-20:
Also stuck on this...