carla-simulator/carla #4988: When i was export a map from carla(use "make package" command) had errors!

huhaocong, 2021-12-16:
CARLA version: 0.9.11
Platform/OS: Ubuntu 20.04

Today i wanted export a map(the map i created), so i followed the guide:
[https://carla.readthedocs.io/en/0.9.11/tuto_A_create_standalone/](url)

And there were some errors happened, like below:
...
[2021.12.16-03.13.09:378][  0]LogUObjectHash: Compacting FUObjectHashTables data took   1.45ms
[2021.12.16-03.13.09:389][  0]LogFileHelpers: No new changes to save!
[2021.12.16-03.13.09:389][  0]LogCore: Engine exit requested (reason: Commandlet PrepareAssetsForCookingCommandlet_0 finished execution (result 0))
1111[2021.12.16-03.13.09:389][  0]LogInit: Display: 
[2021.12.16-03.13.09:389][  0]LogInit: Display: 
[2021.12.16-03.13.09:389][  0]LogInit: Display: Warning/Error Summary (Unique only)
[2021.12.16-03.13.09:389][  0]LogInit: Display: Warning/Error Summary (Unique only)
[2021.12.16-03.13.09:389][  0]LogInit: Display: -----------------------------------
[2021.12.16-03.13.09:389][  0]LogInit: Display: -----------------------------------
1001[2021.12.16-03.13.09:389][  0]LogInit: Display: LogTemp: Error: Package json file not found.
[2021.12.16-03.13.09:389][  0]LogInit: Display: LogTemp: Error: Package json file not found.
1101[2021.12.16-03.13.09:389][  0]LogInit: Display: LogConsoleManager: Warning: Setting the console variable 'r.Streaming.PoolSize' with 'SetByScalability' was ignored as it is lower priority than the previous 'SetByProjectSetting'. Value remains '4000'
[2021.12.16-03.13.09:389][  0]LogInit: Display: LogConsoleManager: Warning: Setting the console variable 'r.Streaming.PoolSize' with 'SetByScalability' was ignored as it is lower priority than the previous 'SetByProjectSetting'. Value remains '4000'
[2021.12.16-03.13.09:389][  0]LogInit: Display: LogFileManager: Warning: ReadFile failed: Count=0 Length=114688 Error=errno=21 (Is a directory) for file 
[2021.12.16-03.13.09:389][  0]LogInit: Display: LogFileManager: Warning: ReadFile failed: Count=0 Length=114688 Error=errno=21 (Is a directory) for file 
[2021.12.16-03.13.09:389][  0]LogInit: Display: 
[2021.12.16-03.13.09:389][  0]LogInit: Display: 
1001[2021.12.16-03.13.09:389][  0]LogInit: Display: Failure - 2 error(s), 3 warning(s)
[2021.12.16-03.13.09:389][  0]LogInit: Display: Failure - 2 error(s), 3 warning(s)
...
[error_info.txt](https://github.com/carla-simulator/carla/files/7724512/error_info.txt)

So, how can i export a map from carla build?
Hope you can help me!

zengxianming, 2022-08-13:
have you solve it? I found the same error. :(

TayYim, 2022-08-31:
For example, demo.xodr and demo.fbx are put in the Import folder.
Specify the package name when doing the import:
`make import  ARGS="--package=demo"`
A 'demo.json' will be generated in the Import folder. If you don't specify the package name, there won't be a 'demo.json' file.

Then make package:
`make package ARGS="--packages=demo"`

It works for me.

ronyshaji, 2022-12-15:
> 

Hey did you exported the default carla map or your own map ?