carla-simulator/carla #4995: Python script's Segmentation fault (core dumped)

DianweiCHEN, 2021-12-19:
CARLA version:0.9.10.1
Platform/OS:Linux
Problem you have experienced: After run the Python script, it didn't work
Hi, I used the ubuntu 20.1 and the Carla 0.9.10.1 and because the former problem which is "No module named 'carla' " 
So I set in the ./bashrc with 
export CARLA_ROOT=/home/andy/Desktop/CARLA_0.9.10.1
export CARLA_SERVER=/home/andy/Desktop/CARLA_0.9.10.1/CarlaUE4.sh 
export PYTHONPATH="$PYTHONPATH:/home/andy/Desktop/CARLA_0.9.10.1/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg"

And after that when I want to run the manual_control.py, it just shows the 'Segmentation fault (core dumped)' in the next line and the carla server is fine. Does anyone know what happend, and thanks!

LucianYuan, 2022-01-14:
I encounter the same segmentation fault, mine is when running the line world.get_random_location_from_navigation(). Also, I am using py3.9 and this issue disappear when I switch to py2.7, can anyone help?

The error is:

Thread 1 "python" received signal SIGSEGV, Segmentation fault.
_PyObject_GC_Alloc (use_calloc=0, basicsize=88)
    at /tmp/build/80754af9/python-split_1631797238431/work/Modules/gcmodule.c:2226
2226	/tmp/build/80754af9/python-split_1631797238431/work/Modules/gcmodule.c: No such directory or file.

RehabHsn, 2022-02-03:
same here I got segmentation fault with that error 

#0  carla::traffic_manager::SimpleWaypoint::GetLocation (this=0x16a6030)
    at ../../LibCarla/source/carla/trafficmanager/SimpleWaypoint.cpp:48
#1  0x00007ffff583b441 in carla::traffic_manager::InMemoryMap::SetUpSpatialTree
    (this=<optimized out>)
    at ../../LibCarla/source/carla/trafficmanager/InMemoryMap.cpp:363
#2  0x00007ffff583ae55 in carla::traffic_manager::InMemoryMap::Load (
    this=0x15fc7f0, 
    content=<error reading variable: Cannot access memory at address 0x8>)
    at ../../LibCarla/source/carla/trafficmanager/InMemoryMap.cpp:161
#3  0x00007ffff585afee in carla::traffic_manager::TrafficManagerLocal::SetupLocalMap (this=0xff4b10)
    at ../../LibCarla/source/carla/trafficmanager/TrafficManagerLocal.cpp:125
#4  0x00007ffff585a733 in carla::traffic_manager::TrafficManagerLocal::TrafficManagerLocal (this=0xff4b10, longitudinal_PID_parameters=..., 
    longitudinal_highway_PID_parameters=..., lateral_PID_parameters=..., 
    lateral_highway_PID_parameters=..., 
    perc_difference_from_limit=<optimized out>, episode_proxy=..., 
    RPCportTM=<optimized out>)
    at ../../LibCarla/source/carla/trafficmanager/TrafficManagerLocal.cpp:107
#5  0x00007ffff5858ed7 in carla::traffic_manager::TrafficManager::CreateTrafficManagerServer (this=<optimized out>, episode_proxy=..., port=<optimized out>)
    at ../../LibCarla/source/carla/trafficmanager/TrafficManager.cpp:148
#6  0x00007ffff5858702 in carla::traffic_manager::TrafficManager::TrafficManager--Type <RET> for more, q to quit, c to continue without paging--c
 (this=<optimized out>, episode_proxy=..., port=<optimized out>) at ../../LibCarla/source/carla/trafficmanager/TrafficManager.cpp:40
#7  0x00007ffff55fa377 in carla::client::Client::GetInstanceTM (this=<optimized out>, port=41856) at dependencies/include/carla/client/Client.h:95
#8  0x00007ffff55dd5c6 in export_client()::$_43::operator()(carla::client::Client const&, unsigned short) const (this=<optimized out>, self=..., t1=41856) at source/libcarla/Client.cpp:210
#9  export_client()::$_43::__invoke(carla::client::Client const&, unsigned short) (self=..., t1=41856) at source/libcarla/Client.cpp:210
#10 0x00007ffff565204c in boost::python::detail::invoke<boost::python::to_python_value<carla::traffic_manager::TrafficManager const&>, carla::traffic_manager::TrafficManager (*)(carla::client::Client const&, unsigned short), boost::python::arg_from_python<carla::client::Client const&>, boost::python::arg_from_python<unsigned short> > (rc=..., f=<optimized out>, ac0=..., ac1=...) at dependencies/include/system/boost/python/detail/invoke.hpp:73
#11 boost::python::detail::caller_arity<2u>::impl<carla::traffic_manager::TrafficManager (*)(carla::client::Client const&, unsigned short), boost::python::default_call_policies, boost::mpl::vector3<carla::traffic_manager::TrafficManager, carla::client::Client const&, unsigned short> >::operator() (this=<optimized out>, args_=0x7fffca51e140) at dependencies/include/system/boost/python/detail/caller.hpp:233
#12 0x00007ffff61e7ea7 in boost::python::objects::function::call(_object*, _object*) const () from /home/oem/.cache/Python-Eggs/carla-0.9.13-py3.8-linux-x86_64.egg-tmp/carla/libcarla.cpython-38-x86_64-linux-gnu.so
#13 0x00007ffff61e9895 in boost::detail::function::void_function_ref_invoker0<boost::python::objects::(anonymous namespace)::bind_return, void>::invoke(boost::detail::function::function_buffer&) () from /home/oem/.cache/Python-Eggs/carla-0.9.13-py3.8-linux-x86_64.egg-tmp/carla/libcarla.cpython-38-x86_64-linux-gnu.so
#14 0x00007ffff61ef522 in boost::python::detail::exception_handler::operator()(boost::function0<void> const&) const () from /home/oem/.cache/Python-Eggs/carla-0.9.13-py3.8-linux-x86_64.egg-tmp/carla/libcarla.cpython-38-x86_64-linux-gnu.so
#15 0x00007ffff5680cb3 in boost::python::detail::translate_exception<rpc::rpc_error, void (*)(rpc::rpc_error const&)>::operator() (this=<optimized out>, handler=..., f=..., translate=0x7ffff55b63b0 <translator(rpc::rpc_error const&)>) at dependencies/include/system/boost/python/detail/translate_exception.hpp:46
#16 boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::_bi::value<void (*)(rpc::rpc_error const&)> >::operator()<bool, boost::python::detail::translate_exception<rpc::rpc_error, void (*)(rpc::rpc_error const&)>, boost::_bi::rrlist2<boost::python::detail::exception_handler const&, boost::function0<void> const&> > (this=<optimized out>, f=..., a=...) at dependencies/include/system/boost/bind/bind.hpp:388
#17 boost::_bi::bind_t<bool, boost::python::detail::translate_exception<rpc::rpc_error, void (*)(rpc::rpc_error const&)>, boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::_bi::value<void (*)(rpc::rpc_error const&)> > >::operator()<boost::python::detail::exception_handler const&, boost::function0<void> const&> (this=<optimized out>, a1=..., a2=...) at dependencies/include/system/boost/bind/bind.hpp:1318
#18 boost::detail::function::function_obj_invoker2<boost::_bi::bind_t<bool, boost::python::detail::translate_exception<rpc::rpc_error, void (*)(rpc::rpc_error const&)>, boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::_bi::value<void (*)(rpc::rpc_error const&)> > >, bool, boost::python::detail::exception_handler const&, boost::function0<void> const&>::invoke (function_obj_ptr=..., a0=..., a1=...) at dependencies/include/system/boost/function/function_template.hpp:137
[... cut ...]
#27 0x000000000068db17 in PyEval_EvalCode ()
#28 0x000000000067eeb1 in ?? ()
#29 0x000000000067ef2f in ?? ()
#30 0x000000000067efd1 in ?? ()
#31 0x000000000067f377 in PyRun_SimpleFileExFlags ()
#32 0x00000000006b7902 in Py_RunMain ()
#33 0x00000000006b7c8d in Py_BytesMain ()
#34 0x00007ffff7de40b3 in __libc_start_main (main=0x4eed30 <main>, argc=2, argv=0x7fffffffdf28, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdf18) at ../csu/libc-start.c:308
#35 0x00000000005fb12e in _start ()

DianweiCHEN, 2022-02-08:
> 

Hi, I found my problem is caused by Pygame. But after I re-install the p=Pygame, this problem is still not fixed. I am stuck currently. Have you fixed your problem yet? Thx

DianweiCHEN, 2022-02-08:
@LucianYuan Hi, I found my problem is caused by Pygame. But after I re-install the p=Pygame, this problem is still not fixed. I am stuck currently. Have you fixed your problem yet? Thx

StephanieMussi, 2022-09-09:
If you are using an imported custom map, you can copy the map folder from /Content to /Content/Carla/Maps. This solves the problem for me.

MahirGulzar, 2022-11-25:
> If you are using an imported custom map, you can copy the map folder from /Content to /Content/Carla/Maps. This solves the problem for me.

 Worked for me. Just to add I am using a custom map  on which I ran `generate_traffic.py` script and was getting Segmentation fault

Here's what I did.

The files/subfiles of the following folder
`CarlaUE4/Content/<custom_map_folder>`
should be copied to their respective folders here

`CarlaUE4/Content/Carla`

For example
- `CarlaUE4/Content/<custom_map_name>/Config/<custom_map_name>.Package.json` 
Goes here ---> `CarlaUE4/Content/Carla/Config`

Similarly,
- `CarlaUE4/Content/<custom_map_name>/Maps/<custom_map_name>/Nav/<custom_map_name>.bin`
Goes here----> `CarlaUE4/Content/Carla/Maps/Nav`

Make sure you do this for all the relevant map files.

schmittlema, 2023-02-27:
@LucianYuan I switched to python3.8 and the problem was resolved