carla-simulator/carla #5003: Is it possible to subdivide the vehicle and pedestrian classes?

soo4826, 2021-12-20:


closed 2022-02-03 by soo4826

tianqi-wang1996, 2022-05-13:
@soo4826 Hello, have you managed to achieve this function? Since I am also looking for a solution.

Best

soo4826, 2022-05-13:
Yes.
I modify some code for subdivide semantic class

There is guideline in this repo: https://github.com/carla-simulator/carla/issues/4064#issue-851601056

Ask me question if you have problem!

tianqi-wang1996, 2022-05-28:
@soo4826 
Hello, could you please tell me how to reorganize the newly created folder as well as the original Vehicles folder.

I am thinking of separating some vehicles from the Vehicles folder to the newly created Vans, Trucks, Motorcycles, and Bicycles folder.

Which files should I include in the new folders apart from the ones named as the vehicle types?  Is it necessary to keep the folder structure the same as the Vehicles folder, e.g. put all the files in the 2Wheeled or 4Wheeled folder? (I have tried to do this, the compiling process succeeded but whenever I spawn such a vehicle, the simulator crashes.)

[image]

soo4826, 2022-05-28:
Hi

In my purpose, I separate 2 wheeled vehicle into motorcycle(PTW, Powered Two Wheeler), Bicycle). 
And separate 4 wheeled vehicle into Commercial vehicle and Vehicle

I do not create new folder, I hard-coded to recognize each class.

Heres's my Tagger.h file.

Also,I dont't and new object into simulator, i just subdivide Vehicle class into (PTW, Bicycle, Commercial Vehicle, Veicle)

If you do no solve your problem, I'll opn my private repo for subdividing vehicle class!

Feel free to ask your question!!

Thanks.

tianqi-wang1996, 2022-05-29:
> 

@soo4826 
Hello, can you share the screen capture of your Vehicles folder, and the new folders inside it, such as the Bicycle folder? Did you just delete the 2wheeled and 4wheeled folder?

tianqi-wang1996, 2022-05-29:
> Hi
> 
> In my purpose, I separate 2 wheeled vehicle into motorcycle(PTW, Powered Two Wheeler), Bicycle). And separate 4 wheeled vehicle into Commercial vehicle and Vehicle
> 
> I do not create new folder, I hard-coded to recognize each class.
> 
> Heres's my Tagger.h file.
> 
> Also,I don't and new object into simulator, i just subdivide Vehicle class into (PTW, Bicycle, Commercial Vehicle, Veicle)
> 
> If you do no solve your problem, I'll opn my private repo for subdividing vehicle class!
> 
> Feel free to ask your question!!
> 
> Thanks.

@soo4826 
I am sorry but I cannot see your tagger.h file, I guess I need to make some change in both Tagger.h and Tagger.cpp, right?

My current folder structure:
[image]

[image]

[image]

In Tagger.h: (how should I code it?)
[image]

In Tagger.cpp:
[image]

Eisoc, 2023-08-26:
@soo4826 @tianqi-wang1996 
Sorry, but want to ask the solution about the issue.
If i changed the folder structure, for example if i move one mesh from 4-wheeled to another folder, it wll crash when spawning it using ros-bridge.
So what is the folder structure and what do i need to do with tagger.h?