carla-simulator/carla #5004: An error occurred when adding a small size vehicle

heshanchuan, 2021-12-21:
CARLA version: 0.9.13
Platform/OS: ubuntu18.04

Hello!

I want to add a self-made car to CARLA. Its size is less than 1m. 
I successfully made a car model using Blender3.0 and added it to CARLA according to the official document. 
But I found that the car will tilt forward while the simulation is running. 
However, when I add a slightly larger vehicle or a full-size vehicle in the same way, there is no such problem. 

Has anyone encountered this problem?

[image]
[image]

pushkar4501, 2022-02-10:
Hi @heshanchuan, i am also looking to add some custom objects in the carla town map using unreal engine editor and make it run using using the pyhton api. 
Could you provide me your steps how did you add the vehicle and run the simulation. 
thank you

LukaszSobczakxBerry, 2022-03-13:
Hi @heshanchuan, have you resolved the issue with a small size vehicle? I'm also using Carla 0.9.13 and I'm trying to add a vehicle with a length of ~70 cm. I found out the same issue as you - my vehicle tilt forward and it can't be steering.
When I'm following the tutorial from [here](https://www.youtube.com/watch?v=w70w1o3nddg), everything works well. But when I try to use my small model than it fails.
[image]

OmarElabd, 2022-04-26:
My testing has shown that the wheel shape radius in the wheel's blueprint must be larger than 18 cm. I think this might be an unreal issue not a carla issue.

HMX2013, 2025-03-30:
I have the same problem. Have you solved it now?

heshanchuan, 2025-03-30:
> I have the same problem. Have you solved it now?
I didn't solve it at the time. Maybe, you can try this method：make the wheels bigger
https://github.com/carla-simulator/carla/issues/5004#issuecomment-1109971940

HMX2013, 2025-03-31:
> My testing has shown that the wheel shape radius in the wheel's blueprint must be larger than 18 cm. I think this might be an unreal issue not a carla issue.

Is it possible to change this limit? The wheel radius of my custom car is just 10.5cm. It can not run normally in Carla UE4.

Blyron, 2025-04-03:
We are attending the issue in https://github.com/carla-simulator/carla/issues/8808

closed 2025-04-03 by Blyron