carla-simulator/carla #5007: Carla-UE4 crashed when i set 'Python Editor Script Plugin' Enabled！

huhaocong, 2021-12-23:
My envirment：
Carla 0.9.11
UE 4.24.3
Ubuntu 20.04

UE4‘s menu step is：Edit->Plugins->Scripting

I want use this plugin to edit some actor's addtributes by python-script, but after i setted this plugin enabled, and restart UE4, it could not start again, and gave me the error message like that:

"Plugin 'PythonScriptPlugin' failed to load because module ''PythonScriptPluginPreload' could not be find.Please ensure the plugin is properly installed,otherwise consider disabling the plugin for this project.

How can i do to set the plugin enabled?

E-Dreamer-LQ, 2022-03-14:
same problem, did you solve it ?

huhaocong, 2022-03-22:
> same problem, did you solve it ?

huhaocong, 2022-03-22:
> same problem, did you solve it ?
No，i think this question of carla is a bug.
Because i changed carla version to 0.9.13,and it's UE4's version is 4.26.2.This function could be useful!!!

closed 2022-03-22 by huhaocong

leochen-xie, 2025-01-10:
> > same problem, did you solve it ?同样的问题，你解决了吗？
> > @E-Dreamer-LQ
> 
> No，i think this question of carla is a bug.不，我认为卡拉的这个问题是一个错误。 Because i changed carla version to 0.9.13,and it's UE4's version is 4.26.2.This function could be useful!!!因为我把carla版本改成了0.9.13，而UE4的版本是4.26.2。这个功能可能有用！

Hello, my Carla version is 0.9.13, but I still have this problem. Have you solved it?