carla-simulator/carla #5014: Is BlueprintLibrary::Find method design error？（C++）

lowbees, 2021-12-30:
see the code

`
client = new carla::client::Client("localhost", 2000);
client->SetTimeout(carla::time_duration::seconds(10));
auto world = client->GetWorld();
auto blueprintLibrary = world.GetBlueprintLibrary();

auto bp = blueprintLibrary->Find("sensor.camera.rgb");
bp->SetAttribute("image_size_x", "1920");    // error, because Find method return const_pointer, I can't use setter

/*
auto bp = blueprintLibrary->at("sensor.camera.rgb");
bp.SetAttribute("image_size_x", "1920");     // ok
*/
`

goksanisil23, 2022-01-22:
I agree that the const_pointer limits the usage.,
As a workaround, you can try:

`bp = (*(blueprintLibrary->Filter("sensor.camera.rgb")))[0];
 bp.SetAttribute("image_size_x", "1920");`

lowbees, 2022-01-28:
> 

Yes, I use Filter method

closed 2023-06-02 by stale[bot]