carla-simulator/carla #5046: Creating a new semantic tag does not work in Windows build version.

gkhandavalli, 2022-01-04:
Hello,
I am currently trying to create a new semantic tag for a custom object.
OS: Windows
Type: Build version
Unreal Editor: Version: 4.26.2-0+++UE4+Release-4.26
CARLA: Latest as of 04 Jan 2022

I followed the steps as described [here](https://carla.readthedocs.io/en/0.9.13/tuto_D_create_semantic_tags/).
I have included a table as the new object for which the tag "Test" with color (RGB: 0,0,70) shall be assigned. I have attached the files I used in this process.

However, when I generate a semantic segmentation image, the table is assigned black color.

[image]

[carla_issue.zip](https://github.com/carla-simulator/carla/files/7810892/carla_issue.zip)
[71-table.zip](https://github.com/carla-simulator/carla/files/7810923/71-table.zip)

Any suggestions will be greatly helpful.

Thank you.

cbushomitre, 2022-01-24:
I am having the same issues with Windows using the latest version of CARLA. My object with a new tag (modified stop sign on the left) also appears as black (0,0,0) even though I created it as (255,0,0) according to the docs. Normal traffic sign (yellow) is on the right
[image]
.

sissiok, 2022-02-03:
I see the same issue on my Ubuntu machine with CARLA 0.9.13 Linux version

closed 2023-06-02 by stale[bot]

TzabarDolev, 2024-01-18:
I still haven't found a solution for it, but the problem is a bit weirder.
It seems that we need to somewhere adjust the number of semantic tags, and i guess that its hard coded on 28 or something like that.
Thing is, once you add another semantic tag, it just uses the one for tag 0 (0, 0, 0), if you'll add another one it will get the color of tag number 1 and so on.
I'm still working on finding a way to change it, but this is what i found out so far.