carla-simulator/carla #5054: generate data in KITTI  format

husamhamu, 2022-01-09:
I wonder if someone has worked on generating detection and tracking data in KITTI dataset format using 0.9.x for Monocular 3d objection detection.
Is there any code that helps collect data ?
 
thanks in advance

husamhamu, 2022-01-11:
So far I saw this pull request [https://github.com/carla-simulator/carla/pull/4992](url) and this file: [https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/client_bounding_boxes.py](url) that creates 3d bounding boxes that are not quite synchronous
but I still don't know how to make use out of these. 

Is it like no one is interested in the topic ?

makaveli10, 2022-03-24:
@husamhamu I tried using this to create a similar tool for latest carla version. As you said not quite synchronous is the issue that I am facing. Did yo make any progress?

husamhamu, 2022-03-24:
@makaveli10 Actually no, I didn't have enough time, so I had to move on and use the 0.8 version of CARLA. But take a look at this repo: [https://github.com/AlanNaoto/carla-dataset-runner](url) it works with the latest version as well, and he is able of saving 2d annotations accurately. So maybe if you would actually save the 3D bounding box information and visualize them later (not on the runtime) you would probably see that 3d box is accurate, and the problem only happens when you try to visualize them on fly like they did in client_bounding_boxes.py. But as I said, I didn't try it.

makaveli10, 2022-03-28:
@husamhamu I did try to save the 3d bbox information in kitti format i.e. ```bounding_box.extent```, ```relative_rotation_y```, ```vehicle.get_transform().location``` as saved in https://github.com/Ozzyz/carla-data-export/blob/26c0bec203a2f3d370ff8373ca6371b7eef35300/bounding_box.py . but it turns out that when trying to build the bounding boxes from saved info it just doesnt work out. Any pointers or resources that you might be able to share?

makaveli10, 2022-03-28:
https://github.com/carla-simulator/carla/issues/5292

Sondosmohamed1, 2022-12-31:
> @husamhamu I did try to save the 3d bbox information in kitti format i.e. `bounding_box.extent`, `relative_rotation_y`, `vehicle.get_transform().location` as saved in [...]

hello  i have the same problem when i render the result it is different,do you find a solution ?