carla-simulator/carla #5057: Large Map import on 0.9.12

aadityasp, 2022-01-11:
CARLA version: 0.9.12 built from source
Platform/OS: Ubuntu 18.04

Hi, I'm trying to import a huge custom map(20kms X 9 kms) from roadrunner. When I follow the steps mentioned at " https://carla.readthedocs.io/en/0.9.12/large_map_import/#making-the-import ", It creates a new folder(with package name) in the Content folder with config,maps and static folders in it. However, the .umap file being generated is empty. It does not have any tiles in it. It just has sky, LargeMapManager, and player start (attaching a screenshot below).  The " make import " command did not throw any error either. So I'm confused. Am I missing something here?  

[image]
How do I load this large map on to Unreal Engine? 

[image]

Any help is greatly appreciated. Thanks!!

zjh6252007, 2022-09-20:
Hello, have you solved this problem?

kharegi, 2022-11-16:
I'm dealing with the same problem.

FBanani, 2023-06-28:
> CARLA version: 0.9.12 built from source Platform/OS: Ubuntu 18.04
>
> Hi, I'm trying to import a huge custom map(20kms X 9 kms) from roadrunner. When I follow the steps mentioned at " [...]

Hi, do you solve this issue? Could you help me?

FBanani, 2023-06-28:
> Hello, have you solved this problem?

Hi, do you solve this issue? Could you help me?

FBanani, 2023-06-28:
> I'm dealing with the same problem.

Hi, do you solve this issue? Could you help me?

> Remember, contributions to this repository should follow our [GitHub Community Guidelines](https://docs.github.com/articles/github-community-guidelines).

Hi, do you solve this issue? Could you help me?

Blyron, 2024-05-10:
If you check sublevels and load them your map should be there. To open sublevels windows follow UE4 guide:
https://docs.unrealengine.com/4.27/en-US/Basics/Levels/LevelsWindow/

Blyron, 2024-05-10:
If this solves your issue, please feel free to close the issue.