carla-simulator/carla #5063: Error running Carla Headless on A100 server

amancodeblast, 2022-01-12:
CARLA version: 0.9.12
Platform/OS: Ubuntu 20.04, Nvidia A100 server cluster
The problem : Not able to run Carla using the `./CarlaUE4.sh -RenderOffScreen` command 

the output :The simulator does not exit but gives a warning
`WARNING: lavapipe is not a conformant vulkan implementation, testing use only.`
Now we tried changing the Town using config.py 
```
python3 config.py -m Town01
load map 'Town01'.
time-out of 20000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
```
The error does not go even after further increasing the Timeout. 
Expectations: Was expecting the headless version to work seamlessly. 
Steps to reproduce: Install Carla using `sudo apt-get install carla-simulator ` on the  A100 following the [documentation](https://carla.readthedocs.io/en/latest/start_quickstart/)

I have not installed UE4 as I don't require map rendering. 
Is there any specific step that we are missing?
Any help is appreciated.  
Thanks

closed 2023-06-02 by stale[bot]

xswings, 2025-02-24:
Execuse me, have you resolve this problem? I thought it is caused by A100，some document said that A100 do not support vulkan and OpenGL，but I‘am not sure.

kkaytekin, 2025-09-25:
Same question as @xswings . Did anyone have a chance to try it out on A100?