carla-simulator/carla #5064: CARLA 0.9.6 with Apollo 5.0 Contatiner issue

marcusvinicius178, 2022-01-12:
CARLA version: 0.9.6
Platform/OS: Ubuntu 20.04
Problem you have experienced: Hi I am following this tutorial to set up CARLA with Apollo 5.0: https://github.com/AuroAi/carla_apollo_bridge. In "USAGE" step, I have an error after run this script inside the carla-apollo container:

`python examples/manual_control.py`

The Error:

`libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  101
  Current serial number in output stream:  102
`

What you expected to happen: I expected the simulator would be launched (because for some reason when I issue this command line outside the carla-apollo container, the simulator opens...however using outside docker, I cannot have communication with Apollo Stack which is running in Apollo container....and no data is received by Dreamview.

Steps to reproduce: I have runner all the steps from this tutorial successfully. And got stuck after running the step:
"**Create an ego vehicle and client**"

Other information (documentation you consulted, workarounds you tried): https://github.com/AuroAi/carla_apollo_bridge

My nvidia driver is ok....I have already runner Apollo.Auto and Autoware.Ai & Autoware.Auto along LGsvl simulator with no issues on nvidia driver or GPU resources, also CARLA is running outside the container...
Therefore I believe that some command regarding Docker or flags could do the trick (**Is this a communication issue between Carla-Apollo container and the CARLA simulator?)**

-->

ldh2020, 2022-01-18:
https://github.com/AuroAi/carla_apollo_bridge/issues/11
Here can solve your problem.

marcusvinicius178, 2022-01-18:
Hi @idh2020 thanks very much for your attention. I have already checked this solution previously. However the Docker commands suggested by this guy, did not work on my system. He suggested here https://github.com/SoonminHwang/dockers/issues/1#issue-350292669, to run the carla-server container using:

docker run -it \
    -v /tmp/.X11-unix:/tmp/.X11-unix:rw \\
    --privileged \
    -e DISPLAY=$DISPLAY \
    -v /usr/lib/nvidia-384:/usr/lib/nvidia-384 \
    -v /usr/lib32/nvidia-384:/usr/lib32/nvidia-384 \
    --device /dev/dri \
    YOUR_DOCKER_IMAGE_NAME

And later inside the docker image:

export PATH="/usr/lib/nvidia-384/bin":${PATH}
export LD_LIBRARY_PATH="/usr/lib/nvidia-384:/usr/lib32/nvidia-384":${LD_LIBRARY_PATH}

**In my case I am using nvidia-470** and in the folder /usr/lib I have the folder "nvidia" not the folder "nvidia-470"....
Then I have modified the command above to my case:

**Command 1**

docker run -it \
    -v /tmp/.X11-unix:/tmp/.X11-unix:rw \\
    --privileged \
    -e DISPLAY=$DISPLAY \
    -v /usr/lib/nvidia:/usr/lib/nvidia \
    -v /usr/lib32/nvidia:/usr/lib32/nvidia \
    --device /dev/dri \
    carla-server

**Command 2**

export PATH="/usr/lib/nvidia/bin":${PATH}
export LD_LIBRARY_PATH="/usr/lib/nvidia/usr/lib32/nvidia":${LD_LIBRARY_PATH}

But after issue on terminal the **Command 1** I got this error:

[image]

I am not an expert in Docker. Would you have some hint to fix this command and **adjust it to my machine** I think in this way I will be available to fix this issue.

ldh2020, 2022-01-19:
> Hi @IDH2020 thanks very much for your attention. I have already checked this solution previously. However the Docker commands suggested by this guy, did not work on my system. He suggested here [SoonminHwang/dockers#1 (comment)](https://github.com/SoonminHwang/dockers/issues/1#issue-350292669), to run the carla-server container using:
> 
> docker run -it -v /tmp/.X11-unix:/tmp/.X11-unix:rw \ --privileged -e DISPLAY=$DISPLAY -v /usr/lib/nvidia-384:/usr/lib/nvidia-384 -v /usr/lib32/nvidia-384:/usr/lib32/nvidia-384 --device /dev/dri YOUR_DOCKER_IMAGE_NAME
> 
> And later inside the docker image:
> 
> export PATH="/usr/lib/nvidia-384/bin":${PATH} export LD_LIBRARY_PATH="/usr/lib/nvidia-384:/usr/lib32/nvidia-384":${LD_LIBRARY_PATH}
> 
> **In my case I am using nvidia-470** and in the folder /usr/lib I have the folder "nvidia" not the folder "nvidia-470".... Then I have modified the command above to my case:
> 
> **Command 1**
> 
> docker run -it -v /tmp/.X11-unix:/tmp/.X11-unix:rw \ --privileged -e DISPLAY=$DISPLAY -v /usr/lib/nvidia:/usr/lib/nvidia -v /usr/lib32/nvidia:/usr/lib32/nvidia --device /dev/dri carla-server
> 
> **Command 2**
> 
> export PATH="/usr/lib/nvidia/bin":${PATH} export LD_LIBRARY_PATH="/usr/lib/nvidia/usr/lib32/nvidia":${LD_LIBRARY_PATH}
> 
> But after issue on terminal the **Command 1** I got this error:
> 
> [image]
> 
> I am not an expert in Docker. Would you have some hint to fix this command and **adjust it to my machine** I think in this way I will be available to fix this issue.

I tested the method with nvidia-410 was OK, but I didn't know 470 how to do. Can you launch manual_control.py in local instead of in the container? If you didn't launch manual_control.py in the container, there won't be this problem.

ldh2020, 2022-01-19:
> Hi @IDH2020 thanks very much for your attention. I have already checked this solution previously. However the Docker commands suggested by this guy, did not work on my system. He suggested here [...]

I tested the method with nvidia-410 was OK, but I didn't know 470 how to do. Can you launch manual_control.py in local instead of in the container? If you didn't launch manual_control.py in the container, there won't be this problem. But, as shown in the picture, you cannot launch the container. Is it?

marcusvinicius178, 2022-01-19:
Yes, locally the script works, however I cannot work with the simulator locally, because the CARLA 0.9.6 is pretty old to install locally (I tried but the installation is not possible, some old libraries and compilers are not even more available for ubuntu 20.04...I have packages conflicts installation that crashes the O.S) 
Unfortunately Apollo 5.0 stack has a bridge available just with CARLA 0.9.6 using Docker....so locally won't work.

And yes, you are right, the docker commands do not work....I would like to understand what is wrong in these commands and adjust them in order to fix this communication with LibGL issue.

ldh2020, 2022-01-20:
Maybe you can change the version of carla to 0.9.10. I test this is OK, and you have to modify the code of lidar in apollo-bridge because of having some update in lidar of 0.9.10. You can find what different of lidar data between 0.9.10 and 0.9.6 in the manual_control.py.

marcusvinicius178, 2022-03-13:
Hi @ldh2020 would you know How to use Localization instead ground truth? 
My doubt is in the last comment of issue below. However Amy answer. I guess Auro.AI stopped working on This bridge.

https://github.com/AuroAi/carla_apollo_bridge/issues/19