carla-simulator/carla #5066: random_left_lanechange_percentage doesn't work

Miknow78, 2022-01-12:
def main():
    client = carla.Client('localhost', 2000)
    client.set_timeout(60.0)
    world = client.get_world()
    settings = world.get_settings()
    settings.synchronous_mode = True
    settings.fixed_delta_seconds = 0.05
    world.apply_settings(settings)
    traffic_manager = client.get_trafficmanager(8000)
    traffic_manager.set_synchronous_mode(True)

    try:
        actors_list = []

        blueprint_library = world.get_blueprint_library()

        bp = blueprint_library.find('vehicle.mercedes.sprinter')

        spawnPoint = carla.Transform(carla.Location(x=23, y=248, z=0.2), carla.Rotation(pitch=0, yaw=0, roll=0))

        vehicle = world.spawn_actor(bp, spawnPoint)
        vehicle.set_target_velocity(carla.Vector3D(3.94, 0, 0))

        actors_list.append(vehicle)
        vehicle.set_autopilot(True)

        for frame in range(0, 200):
            world.tick()

            if frame == 10:
                traffic_manager.random_left_lanechange_percentage(vehicle,100)

            if frame == 100:
                traffic_manager.random_right_lanechange_percentage(vehicle,100)
  
Why this doesn't work?

jackbart94, 2022-01-13:
It works correctly for me. Can you present a video of your issue?

Miknow78, 2022-01-14:
I changed my code structure (inserted a def main) and it worked.

closed 2022-01-28 by jackbart94