carla-simulator/carla #5069: Upload Carla map To APollo Dreamview

marcusvinicius178, 2022-01-13:
Hi i am using CARLA 0.9.6 with this tutorial: https://github.com/AuroAi/carla_apollo_bridge to set up CARLA with APOLLO.

I have done all the steps successfully however, in the last step ( **to visualize camera data from the simulator in Dreamview**) Apollo does not displays the Carla03Town as in the tutorial
[image]

A zoom:

[image]

In tutorial the City: Town03 appears. 
How do I add this map in Dreamview from Apollo in order to get the images from the simulator?

Thanks in advance

ldh2020, 2022-01-17:
Town06 and Town07 maps are imported successfully. Why is Town03 map failed? The method is the same.

marcusvinicius178, 2022-01-17:
Yes now I am doing in other machine...it worked. I guess that my old machine was using the cache instead pull the new scenarios docker images....because previously I was using LGSVL simulator for other project.

But **now I am facing this issue**:

libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  101
  Current serial number in output stream:  102

when launching the manual_control.py script inside docker..the libGL is not found. I tried numerous workarounds, reinstall nvidia-driver, install mese-drivers, vulkan drivers, add sim links: https://askubuntu.com/questions/541343/problems-with-libgl-fbconfigs-swrast-through-each-update/566522#566522

and also tried this approach: https://github.com/SoonminHwang/dockers/issues/1#issue-350292669:

`docker run -it \
    -v /tmp/.X11-unix:/tmp/.X11-unix:rw \\
    --privileged \
    -e DISPLAY=$DISPLAY \
    -v /usr/lib/nvidia:/usr/lib/nvidia \
    -v /usr/lib32/nvidia:/usr/lib32/nvidia \
    --device /dev/dri \
    carla_apollo

export PATH="/usr/lib/nvidia/bin":${PATH}
export LD_LIBRARY_PATH="/usr/lib/nvidia/usr/lib32/nvidia":${LD_LIBRARY_PATH}`

 
that would fix the issue, however this command did not work in my computer, because I have other nvidia-driver. I have tried to put my nvidia-driver instead...but no success. The command does not work. I am not that good to understand the Docker images flags. Do you know how the command from this git issue could be adjusted to my machine...I am using nvidia-driver 470?

Or how to fix this issue? Did you have this libGL and swrast issue when you used Apollo+CARLA in docker? How to fix it?

I would like a more recent tutorial of CARLA and Apollo do you know if is it available? version 0.9.6 is too old to install locally with ubuntu 20.04

ldh2020, 2022-01-18:
I tested nvidia-410 was OK, but I didn't know 470 how to do. Can you launch manual_control.py in local instead of in the container?

ldh2020, 2022-01-18:
> Yes now I am doing in other machine...it worked. I guess that my old machine was using the cache instead pull the new scenarios docker images....because previously I was using LGSVL simulator for other project.
> 
> But **now I am facing this issue**:
> 
> libGL error: No matching fbConfigs or visuals found libGL error: failed to load driver: swrast X Error of failed request: BadValue (integer parameter out of range for operation) Major opcode of failed request: 152 (GLX) Minor opcode of failed request: 3 (X_GLXCreateContext) Value in failed request: 0x0 Serial number of failed request: 101 Current serial number in output stream: 102
> 
> when launching the manual_control.py script inside docker..the libGL is not found. I tried numerous workarounds, reinstall nvidia-driver, install mese-drivers, vulkan drivers, add sim links: https://askubuntu.com/questions/541343/problems-with-libgl-fbconfigs-swrast-through-each-update/566522#566522
> 
> and also tried this approach: [SoonminHwang/dockers#1 (comment)](https://github.com/SoonminHwang/dockers/issues/1#issue-350292669):
> 
> `docker run -it -v /tmp/.X11-unix:/tmp/.X11-unix:rw \ --privileged -e DISPLAY=$DISPLAY -v /usr/lib/nvidia:/usr/lib/nvidia -v /usr/lib32/nvidia:/usr/lib32/nvidia --device /dev/dri carla_apollo
> 
> export PATH="/usr/lib/nvidia/bin":${PATH} export LD_LIBRARY_PATH="/usr/lib/nvidia/usr/lib32/nvidia":${LD_LIBRARY_PATH}`
> 
> that would fix the issue, however this command did not work in my computer, because I have other nvidia-driver. I have tried to put my nvidia-driver instead...but no success. The command does not work. I am not that good to understand the Docker images flags. Do you know how the command from this git issue could be adjusted to my machine...I am using nvidia-driver 470?
> 
> Or how to fix this issue? Did you have this libGL and swrast issue when you used Apollo+CARLA in docker? How to fix it?
> 
> I would like a more recent tutorial of CARLA and Apollo do you know if is it available? version 0.9.6 is too old to install locally with ubuntu 20.04
> 
> This issue is better detailed here:
> 
> #5064

I tested the method with nvidia-410 was OK, but I didn't know 470 how to do. Can you launch manual_control.py in local instead of in the container?

marcusvinicius178, 2022-01-25:
I am going to close this issue as I have solved the map issues here:https://github.com/ApolloAuto/apollo/issues/14274#issuecomment-1019617353

@ldh2020 yes I could launch this script outside docker. However the other script needed to communicate with Apollo 5.0 cannot be launched outside docker: the "run_bridge.py" script. Because it imports modules which are built inside docker environment and are not available outside docker. I am still struggling with this issue. I have opened a new issue here: https://github.com/AuroAi/carla_apollo_bridge/issues/30 in the proper channel. I will be very grateful if you can help me there 😊

closed 2022-01-25 by marcusvinicius178

reylio06, 2022-03-08:
Hello @marcusvinicius178.
I've been struggling for a while to run carla with Apollo and i've seen you've also struggled with this topic. I was trying to run it locally with a prebuilt version of Carla 0.9.6 and i had the same issue while trying to run the client from carla-apollo-bridge. 
Running the client locally started my apollo dreamview but i had no data for camera view and sensors. 
I wonder if you can actually run it from a local build or how complicated it is to modify the images/containers in order to do so.
Thank you

zhouyu102030, 2022-09-15:
I would like to ask where can I get the map resources of town02?
Thank you.

ldh2020, 2022-09-16:
> I would like to ask where can I get the map resources of town02? Thank you.

Look here.  https://github.com/AuroAi/carla_apollo_bridge

zhouyu102030, 2022-09-21:
> > 请问town02的地图资源在哪里可以得到？谢谢你。
> 
> 看这里。 https://github.com/AuroAi/carla_apollo_bridge

> > Town02 的地图资源在哪里可以得到？谢谢你。
> 
> 看这里。 https://github.com/AuroAi/carla_apollo_bridge

Thank you, I solved the problem.

JasonBourne1998, 2023-02-14:
> > I would like to ask where can I get the map resources of town02? Thank you.
> 
> Look here. https://github.com/AuroAi/carla_apollo_bridge

Hi @ldh2020 @zhouyu102030 , I still can't find the map resource in the your link.