carla-simulator/carla #5070: Retrieving image data as float values

brscholz, 2022-01-14:
Hi there,

I want to retrieve image data from the rgb camera in full float precision. So I created my own pixel type with 4 float channels and a camera with 

Enable16BitFormat(true)
FPixelReader::SendPixelsInRenderThread(*this, true) 

similar to the optical flow camera. When I retrieve the data from buffer as type np.half, I get the strangest array with nan and negative float values. I also wrote a method similar to the creation of the FakeImage in the OpticalFlowCamera, that is supposed to deliver the 8bit representation of my float image, but this one crashes with SegmentationFault.

Hints on what I'm doing wrong would be highly appreciated.

Best regards
Robert

brscholz, 2022-01-14:
Okay, found the mistake by myself.
I suggest renaming the method "WriteFloatPixelsToBuffer_Vulkan" to "WriteFlowValuesToBuffer" as this is what it really does...

IbrahimYang, 2022-03-08:
Hi brscholz,
I'm still struggling with rendering HDR images in Carla. May you tell me your modification is working or not? If yes, may you share this core with us? Thanks！

closed 2022-03-11 by brscholz

manuelmaiorb, 2022-07-07:
Hello! I have a question related to the pixel values. What the range of float values per channel should be? As far as I know, for 16-bit floats, Unreal Engine outputs values between 0.0f - 65535.0f. Does this also apply in your implementation?

brscholz, 2022-07-07:
Hi Manuel, for the HDR camera we transfer float values, not uint_16. That means, our outputs range between 0 and the maximum float value for HDR defined by UE4, I'm not 100% sure about that exakt value. You will need to apply some kind of tone mapping.

manuelmaiorb, 2022-07-08:
Thank you for your answer! Meanwhile I have taken a look at what the range of values is for ETextureRenderTargetFormat::RTF_RGBA16f (which is set for the render target). According to [this resource](https://docs.unrealengine.com/4.26/en-US/API/Runtime/Engine/Engine/ETextureRenderTargetFormat/), even in 16-bit float format, per pixel values can get values up to ~65000. I have tested your implementation and more specifically the problem I have is that if I raise the camera exposure compensation to ~30.0 (which I need to do for the renders which I want to obtain), the pixel values are saturating at exactly (R=100.0625, G=99.9375, B=100.00). Also, for checking the values, I have captured the results by saving images directly from Carla plugin, from the corresponding render target, and not from Python API.

Domsall, 2022-07-11:
A few things are not clear to me:
1. What is wrong with the current implementation? We actually just output the raw values from Unreal (which can be higher than 1.0). There is a default tone mapping on the clients side, but this can be changed. The values from Unreal come from `ReadSurfaceFloatData` and I doubt that those values will reach 2^16.
2. What is your "Scene Capture Component" reference doing? Can you share it? Can you compare the Carla picture with the reference picture? Are the scenes from the same Carla level? Is the tone mapping / post processing the same?

manuelmaiorb, 2022-07-11:
I did not suppose that something is wrong with the implementation. I just expected it to output something different. If you say that the default tone mapping can be changed, then I think I might give a try changing something there. Could you tell me where can I access these tone mapping configurations?

ztysdu, 2022-10-13:
Hi, I'm recently researching how to generate HDR images with carla, and I only found this [URL](https://docs.unrealengine.com/4.26/en-US/RenderingAndGraphics/HDRDisplayOutput/) and your issue on the web. Can you tell me how you set up HDR mode? Thank you very much!