carla-simulator/carla #5074: Traffic Manager Auto Lane Change

Miknow78, 2022-01-14:
What is the criteria for a vehicle to change lane when it's in autopilot mode?
I set traffic_maanger.auto_lane_change(vehicle, True) and a leading distance of 3. 
If my car has a speed of 50 and the car in front has a speed of 20, shouldn't my vehicle change his lane?

tiagorochag, 2022-08-03:
Hi @Miknow78, did you manage to understand how change lane works in autopilot?

I also created a scenario with traffic_manager.auto_lane_change(v, True) where a obstacle vehicle is spawned in front of the 'ego' that is configured with auto lane change as true. But never the vehicle manages to overtake the static obstacle in front of it.
It only changes lanes with the force command as: traffic_manager.force_lane_change(danger_car, False). 

@jackbart94 hello could you please take a look on this?

AreopagX, 2024-04-09:
@germanros1987 @bernatx any news on this?