carla-simulator/carla #5087: Python API reference misses set_path and set_route

j-scharrenbach, 2022-01-19:
Hi everybody,

not sure if this is on purpose or not.  Feel free to close the issue if so.

I was trying to make a car always do a specific maneuver in a scenic scenario. I stumbled accross the `set_path` and `set_route` functionality as definined [here](https://github.com/carla-simulator/carla/blob/a1b37f7f1cf34b0f6f77973c469926ea368d1507/PythonAPI/carla/source/libcarla/TrafficManager.cpp#L109). Both functions are not mentioned in the Python API reference. It would be nice to add it to the documentation as others will find it more easily.

Best regards,
Jannik

albertxos, 2022-01-19:
indeed: 
https://carla.readthedocs.io/en/latest/python_api/#carlatrafficmanager

sschupp, 2022-01-21:
Could you give me a quick tutorial on how to use `set_path` or `set_route`? I have a route that is provided in the same way the `set_destination` creates a route for Agents, but I can't manage to match the C++ Signature

j-scharrenbach, 2022-01-22:
> Could you give me a quick tutorial on how to use `set_path` or `set_route`? I have a route that is provided in the same way the `set_destination` creates a route for Agents, but I can't manage to match the C++ Signature

Hi Steffen,

I only used the `set_route` function, so I only can give you detailed information about this. I use the funktion like this:
```
# get your instance of traffic_manager and the corresponding carla_actor
trafficmanager.set_route(carla_actor, ["Left", "Right", "Straight"])
```
In this case, the actor will turn left on the first intersection, turn right on the second one and then drives straight. Just pass a list with the desired actions to take. You can find all options [here](https://github.com/carla-simulator/carla/blob/a1b37f7f1cf34b0f6f77973c469926ea368d1507/PythonAPI/carla/source/libcarla/TrafficManager.cpp#L34).

I also tried the `set_path` function once but **I don't now for sure if I used it correctly**. As far as I remeber I did the following:
```
from carla.libcarla import Location
# get your instance of traffic_manager and the corresponding carla_actor
path = [Location(x=0., y=1., z=0.), Location(x=3., y=4., z=0.)]   # add all the points of the desired path to a list of Location() objs
traffic_manager.set_path(carla_actor, path)
```

I hope this helps!

glopezdiest, 2022-01-28:
Hey @j-scharrenbach, yeah, the docs are a bit being, as those features were just introduced in 0.9.13, and are working to add them.

As for how they work, everything you explains is correct. However, I'd like to add that those aren't meant to be route planners, but directions to take at a given junction. You can either use `set_path` with a high level command such as "Left", "Straight"... or `set_route` with CARLA locations

2Innovative, 2022-03-16:
I have a problem where the vehicles I spawn do drive by autopilot but don't stick to the applied path, they sometimes even crash into a tree on the sidewalk. 
Why would CARLA do that? 
The code is rather simple I think.

M-Colley, 2026-10-24:
The original missing-reference entries are present on current `ue58-dev` at `defabe35d1ad47c4231d4f7fb1f4e46ac2e08ec5`: [TrafficManager.set_path](https://github.com/carla-simulator/carla/blob/defabe35d1ad47c4231d4f7fb1f4e46ac2e08ec5/PythonAPI/docs/client.yml#L800) and [set_route](https://github.com/carla-simulator/carla/blob/defabe35d1ad47c4231d4f7fb1f4e46ac2e08ec5/PythonAPI/docs/client.yml#L820), including `actor`, `path`, and `empty_buffer` parameters. The entries also appear in the generated Python API reference; I verified the YAML and generated file in a fresh checkout.

This addresses the documentation omission. The separate route-following behavior mentioned later in this thread was not retested, and this check does not establish which hosted documentation version has been deployed.