carla-simulator/carla #5088: Make Launch fail - Carla 0.9.13

Shroculus, 2022-01-20:


LucasFKobernic, 2022-01-20:
Did you set unreal engine environment variable?

because this error here means that it can't find Unreal Engine:

-[BuildCarlaUE4]: Building Unreal Editor...
The system cannot find the path specified.

I had it one time, and by me was the enrinvonrment variable not set. Or it's set wrong.

Shroculus, 2022-01-20:
> Did you set unreal engine environment variable?
> 
> because this error here means that it can't find Unreal Engine:
> 
> -[BuildCarlaUE4]: Building Unreal Editor... The system cannot find the path specified.
> 
> I had it one time, and by me was the enrinvonrment variable not set. Or it's set wrong.

Yes, I did set the Unreal Engine environment variable as mentioned above:
I first set my environment variable UE4_ROOT to C:\UnrealEngine\Engine\Binaries\Win64

LucasFKobernic, 2022-01-20:
I think u just need to give the path to root folder and not to Win64:

try only this:

C:\UnrealEngine\

then should work

Shroculus, 2022-01-20:
> I think u just need to give the path to root folder and not to Win64:
> 
> try only this:
> 
> C:\UnrealEngine\
> 
> then should work

Thank you, that worked. But after 75% of initialization of the editor it crashed with 

"FATAL ERROR" TheUE4-CarlaUE4 Editor has crashed. I guess I will open another issue chain for this.

closed 2022-01-20 by Shroculus

ashukr07, 2024-02-20:
## I am trying to run "make launch" command but everytime this error comes in. Can anyone please help
carla version->0.9.15
OS-> windows 11

-[BuildUE4Plugins]: [Batch params]: --build --all
fatal: unable to access 'https://github.com/carla-simulator/StreetMap.git/': Could not resolve host: github.com
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "C:\carla\Unreal\CarlaUE4\Plugins"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs: 8
-[Setup]: Boost toolset: msvc-14.2
-[Setup]: Generator: "Visual Studio 16 2019"
-[Setup]: Install directory: "C:\carla\Build"
-[Setup]: Installing zlib...
-[install_zlib]: [Batch params]: --build-dir "C:\carla\Build"
-[install_zlib]: A zlib installation already exists.
-[install_zlib]: Delete "C:\carla\Build\zlib-install" if you want to force a rebuild.
-[install_zlib]: Exiting...
1 file(s) copied.
-[Setup]: Installing libpng...
-[install_libpng]: [Batch params]: --build-dir "C:\carla\Build" --zlib-install-dir "C:\carla\Build\zlib-install"
-[install_libpng]: A libpng installation already exists.
-[install_libpng]: Delete "C:\carla\Build\libpng-1.2.37-install" if you want to force a rebuild.
-[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
-[install_rpclib]: [Batch params]: --build-dir "C:\carla\Build" --generator "Visual Studio 16 2019"
-[install_rpclib]: A rpclib installation already exists.
-[install_rpclib]: Delete "C:\carla\Build\rpclib-install" if you want to force a rebuild.
-[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
-[install_gtest]: [Batch params]: --build-dir "C:\carla\Build" --generator "Visual Studio 16 2019"
The system cannot find the batch label specified - already_build

EHaarer, 2024-03-31:
Hello, I too am having a similar problem when attempting to run "make launch" in the x64 Native Tools Command Prompt.
I am running the most recent release on windows 11 and am setting up CARLA for the first time. My environment variable is set both in UE4_ROOT and in PATH as C:\Users\e_haa\Documents\CarlaRepo\Engine where CarlaRepo is where I've installed Unreal Engine following the steps entailed [here](https://carla.readthedocs.io/en/latest/build_windows/). I have also set the environment variable to just C:\Users\e_haa\Documents\CarlaRepo\ to no avail. make PythonAPI ran without a hitch (I am assuming the .egg file entailed in the download is generated during this step? if not the please advise) Otherwise I have followed all other steps as described in the aforementioned link.
Here is the Error:
C:\Users\e_haa\Documents\Carla Sim Git\carla>make launch
-[BuildUE4Plugins]: [Batch params]: --build --all
fatal: Too many arguments.

usage: git clone [<options>] [--] <repo> [<dir>]

    -v, --verbose         be more verbose
    -q, --quiet           be more quiet
    --progress            force progress reporting
    --reject-shallow      don't clone shallow repository
    -n, --no-checkout     don't create a checkout
    --bare                create a bare repository
    --mirror              create a mirror repository (implies bare)
    -l, --local           to clone from a local repository
    --no-hardlinks        don't use local hardlinks, always copy
    -s, --shared          setup as shared repository
    --recurse-submodules[=<pathspec>]
                          initialize submodules in the clone
    --recursive ...       alias of --recurse-submodules
    -j, --jobs <n>        number of submodules cloned in parallel
    --template <template-directory>
                          directory from which templates will be used
    --reference <repo>    reference repository
    --reference-if-able <repo>
                          reference repository
    --dissociate          use --reference only while cloning
    -o, --origin <name>   use <name> instead of 'origin' to track upstream
    -b, --branch <branch>
                          checkout <branch> instead of the remote's HEAD
    -u, --upload-pack <path>
                          path to git-upload-pack on the remote
    --depth <depth>       create a shallow clone of that depth
    --shallow-since <time>
                          create a shallow clone since a specific time
    --shallow-exclude <revision>
                          deepen history of shallow clone, excluding rev
    --single-branch       clone only one branch, HEAD or --branch
    --no-tags             don't clone any tags, and make later fetches not to follow them
    --shallow-submodules  any cloned submodules will be shallow
    --separate-git-dir <gitdir>
                          separate git dir from working tree
    -c, --config <key=value>
                          set config inside the new repository
    --server-option <server-specific>
                          option to transmit
    -4, --ipv4            use IPv4 addresses only
    -6, --ipv6            use IPv6 addresses only
    --filter <args>       object filtering
    --also-filter-submodules
                          apply partial clone filters to submodules
    --remote-submodules   any cloned submodules will use their remote-tracking branch
    --sparse              initialize sparse-checkout file to include only files at root

The system cannot find the path specified.
fatal: reference is not a tree: 260273d6b7c3f28988cda31fd33441de7e272958
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "C:\Users\e_haa\Documents\Carla Sim Git\carla\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  16
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 16 2019"
-[Setup]: Install directory:  "C:\Users\e_haa\Documents\Carla Sim Git\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Users\e_haa\Documents\Carla Sim Git\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\Users\e_haa\Documents\Carla Sim Git\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
The system cannot find the file specified.
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\Users\e_haa\Documents\Carla Sim Git\carla\Build\" --zlib-install-dir "C:\Users\e_haa\Documents\Carla Sim Git\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\Users\e_haa\Documents\Carla Sim Git\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\Users\e_haa\Documents\Carla Sim Git\carla\Build\" --generator "Visual Studio 16 2019"
Sim was unexpected at this time.
make: *** [setup] Error 255